prusa3d/Prusa-Firmware-Buddy #3302: [BUG] Crash when loading file from USB MINI+ 5.1.0

rinkek, 2023-09-22:
When loading file from USB drive, the printer crashes to a blue screen.
There is something with the file, i simply cannot load it, whatever i try.
The exact same gcode works fine when printing via PrusaLink.

I'm printing on a MINI Bear, so my printer code is a bit off.
Attached the crash dump

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12703213/dump_buddy.zip)

rinkek, 2023-09-22:
I think this is because the gcode files are in a subfolder.
I moved them to the root folder, then they print fine.

bg1627, 2023-09-23:
exactly the issue for me, too, and yes that's what you have to do for now, rinkek.  If I move the file to the root, it prints fine.

closed 2024-06-25 by github-actions[bot]