prusa3d/Prusa-Firmware-Buddy #3304: [BUG] Mini+ 5.1.0 alpha  BSOD "Imprecise Data Access"

stefanetzkorn, 2023-09-22:
**Printer type** - [MINI]

**Printer firmware version** - [5.1.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor - E3D Revo Micro Hotend]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Got BSOD while printing

**How to reproduce**
  Print the attached Gcode

**Expected behavior**
  Should not get a BSOD

**G-code**
[raketovy motor XL - high poly II_0.4n_0.2mm_PLA_MINIIS_1h44m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12704154/raketovy.motor.XL.-.high.poly.II_0.4n_0.2mm_PLA_MINIIS_1h44m.gcode.zip)

**Crash dump file**
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12704160/dump_buddy.bin.zip)

**Picture**
[image]

javree, 2023-09-23:
[DrawerBoxes_77x102_0.4n_0.2mm_PLA_MINIIS_2h12m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12705991/DrawerBoxes_77x102_0.4n_0.2mm_PLA_MINIIS_2h12m.gcode.zip)
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12705992/dump_buddy.bin.zip)
[image]
Seeing an identical issue, model printed fine 4 times, 5th time got a crash
The screen looks identical, my model is a lot easier though. Printing in PLA, with almost stock profile

MINI+ , 100% stock , 5.1.0-alpha , printing from USB (known good stick)
Network link was through cable , IPv4 through DHCP
PrusaConnect was active
PrusaLink was active, webpage was opened for status monitoring

Jerzeek, 2023-09-26:
Got the same BSOD today

Jonty, 2023-10-13:
I just experienced the same crash mid-print. Can provide dump_buddy.bin if needed.

closed 2024-06-25 by github-actions[bot]