prusa3d/Prusa-Firmware-Buddy #3308: [BUG] 5.1.0 crashed, unable to go back to 4.4.1

kraut55, 2023-09-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [5.1.0]

**Original or Custom firmware** - [Original/Custom]

**USB drive 

**Describe the bug**
    I tried to update my Mini+ to 5.1.0. System crashed. Tried to go back to 4.4.1. Got error #12602 and #12604. Nothing worked. Endless loop Bootloader 2.1.2 to about 50%. Tried againto upload. Same result. Have a brick sitting on my desk now, unable to print anything.
[image]

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

darraghbr, 2023-09-24:
Have you tried the 'alternate method' described here? I believe you need to be sure there is only one firmware file on the USB stick?

[Prusa article: Firmware updating (MINI/MINI+)](https://help.prusa3d.com/article/firmware-updating-mini-mini_124784)

kraut55, 2023-09-24:
I tried everything:
Formatted stick with MSDOS(FAT32) and only 5.1.0
Formatted again and then the two 4.4.1 files
Then tried each file alone
Repeated the same procedures with another USB stick

darraghbr, 2023-09-24:
As long as you are double clicking the button shortly after pressing the reset button, [as seen in this video here.](https://youtu.be/8OgkHWrGXTM)

If that doesn't work then try contacting Prusa Support, unless you have broken the appendix it may still be covered under warranty.

kraut55, 2023-09-24:
Double clicking never seemed to work. I always had to click several times very fast to get to the Flash bitton.
I tried Prusa Support, spent most of 55 min waiting for a suggestion, instead being asked for my order no (I gave him the serial right at the start). 15 min without any action whatsoever. The end result was that the problem would be dealt with by the technicians and they would inform me by email.
What a difference to the normally competent support. 
„Broken appendix“ what appendix do you mean?

BTW Thanks for yr help

kraut55, 2023-09-24:
I kept on trying getting all the crash reports etc. again and again but it seems I am back now at 4.1.1. going through the setup routine again. Cross my fingers. 
Thanks again for your help.

darraghbr, 2023-09-24:
Well I'm glad you have a printer that works again. How did you fix it in the end?

The appendix is a piece on the control board that needs to be broken off if you want to flash firmware that is not published by Prusa.

kraut55, 2023-09-25:
I tried 5.1.0 again and it crashed. Then I tried 4.4.1 with both files and pressed just twice, this time with normal speed and it loaded the firmware. 
For some reason I get an error on the y-axis but at least the firmware problem is solved.

closed 2023-09-25 by kraut55