prusa3d/Prusa-Firmware-Buddy #3310: [QUESTION] Where is the code responsible for the Input Shape?

R1kk3r, 2023-09-24:
Hi all,
I saw the commit for the release 5.0. Does anyone know where this code comes from? There is no branch nor PR associated with it. Same question for the new Input Shaper, any commit history or documentation about the way it works? Where is the code implementing the feature?

Thank you !

rinkek, 2023-09-24:
Isn't it just the M593 command in Marlin?

murk-sy, 2023-09-25:
Unfortunately commit history was essentially killed off when an unreleased Prusaslicer feature in development was released for a different slicer. Now the actual development (of all projects) is essentially completely separate.

There is some info on IS on https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.0.0-alpha1 but I don't think there's good complete documentation anywhere yet.

bkerler, 2024-02-20:
The input shaper related code can be found [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/b3ee47aa573f4d9b4af82d7a6963f43ab37ccc08/lib/Marlin/Marlin/src/feature/input_shaper)

bkerler, 2024-02-20:
Related GCode M74 is [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/b3ee47aa573f4d9b4af82d7a6963f43ab37ccc08/lib/Marlin/Marlin/src/gcode/feature/input_shaper/M74.cpp#L4)

danopernis, 2024-05-30:
Question already answered by @bkerler

closed 2024-05-30 by danopernis