prusa3d/Prusa-Firmware-Buddy #3311: [BUG] Mini stops printing mid-print with a blue screen - possible fire hazard?

Steinamyte, 2023-09-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - 5.1.0 ALpha1+12820

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Sensor

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
After some flawless prints the days before with the new 5.1.0 alpha Firmware, the Mini now suddenly stopped mid-print with a blue screen (no photo taken but the screen was full of text) and the alarm beeping. I confirmed and could restart it without issues. But the nozzle was in the middle of the part, touching the surface. I am not sure in which state it would have remained in that blue screen situation - worst case would be nozzle stays powered up contacting the surface, creating a possible fire hazard!

**How to reproduce**
Did not try to reprint the same file again yet. 

**Expected behavior**
Machine prints without issues as the days before with 

**G-code**
Attached

**Crash dump file**
  Attached

[tripp-trapp-stokke-chairv4-side-1_0.4n_0.2mm_PLA_MINIIS_1h26m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12709696/tripp-trapp-stokke-chairv4-side-1_0.4n_0.2mm_PLA_MINIIS_1h26m.zip)

tfossey, 2023-09-26:
I have been experiencing multiple crashes simultaniously for various reasons but after I had exactly the same crash as this I took a picture of the screen (attached). When I rebooted I selected a menu item (can't remember what) and it crashed again so took a picture of that, also attached.

[image]

[image]

G-code and crash dump

[Prusa Mini Z-axis support brace, by MaxT_0.4n_0.2mm_PETG_MINIIS_3h25m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12722519/Prusa.Mini.Z-axis.support.brace.by.MaxT_0.4n_0.2mm_PETG_MINIIS_3h25m.gcode.zip)

[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12722507/dump_buddy.bin.zip)

Jerzeek, 2023-09-26:
Same here with the connect task

darraghbr, 2023-09-27:
Is this a duplicate of #3304 ?

NeuralDEEPnet, 2023-09-27:
I believe this is a bug when Cloud Connect makes a temp file, and later it gets deleted before the print ends...

closed 2024-05-30 by danopernis