prusa3d/Prusa-Firmware-Buddy #3312: [BUG] Print head stalls during printing after firmware update to 5.0.0

wagnbtd, 2023-09-25:
**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Original or Custom firmware** - [Original]

**Optional upgrades** - Stock MK4 kit built, Copperhead Diamond nozzle. 

**USB drive** - Samsung MUF-64AB/AM - 3.1 Flash drive formated to 32gb
  
**Describe the bug**
  After upgrading the printer to 5.0.0, sliced gcode file in Prusaslicer 2.6.1, Input Shaper printer profile, 0.2 layer height - Structural. 3 or 4 layers up in the print, the print head extrudes in an unexpected straight line to a spot and just stalls in that spot. Print head and bed stay at temperature. Screen shows half display of print image but I can press the print knob and go into tune and back out. Sometimes this will seem to get the print to resume. Not sure if it is due to time or screen interaction. This has happened twice back to back. Once with PLA, and once with PC Blend. Same STL models both times but different orientations. Organic supports each time. After the stall resumes, print seems to finish fine. 

**How to reproduce**
  Only steps were, slice model, send to printer through prusalink to USB drive, and print. 

**Expected behavior**
  Standard print start to finish

**G-code**
  gcode attached - tpa_bases_pla_0.4n_0.2mm_PLA_MK4IS_6h28m.gcode.txt

[tpa_bases_pla_0.4n_0.2mm_PLA_MK4IS_6h28m.gcode.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12710165/tpa_bases_pla_0.4n_0.2mm_PLA_MK4IS_6h28m.gcode.txt)

Kris5025s, 2023-09-25:
Have you seen this bug: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156 ?
I would check if your gcode wasn't corrupted when sending it via prusalink, just to cross out known problem with prusalink and some of USB drives. I had very similiar symptoms when my gcode was corrupted (extrusions outside the model and weird pauses / slow downs during the print).
Is the gcode file you attached coming from USB drive used in a printer or you resliced the model and attached gcode from your PC?

wagnbtd, 2023-09-26:
> Have you seen this bug: #3156 ? I would check if your gcode wasn't corrupted when sending it via prusalink, just to cross out known problem with prusalink and some of USB drives. I had very similiar symptoms on 4.7.2 firmware when my gcode was corrupted (extrusions outside the model and weird pauses / slow downs during the print). Is the gcode file you attached coming from USB drive used in a printer or you resliced the model and attached gcode from your PC?

Based on a notepad++ side-by-side comparison, I do see differences in the file pushed to the USB over Prusalink. Some lines seem truncated, such as missing values at the end of lines "G1 X56.478 Y141.84 E"  and some at the beginning of others ".662 Y141.642 E.00604". It only seems to happen on larger file copies. Smaller ones I have done since don't seem to have the issue.

closed 2024-05-30 by danopernis