prusa3d/Prusa-Firmware-Buddy #3315: [FEATURE REQUEST] Display "automatic start-up" warning symbol when printer is in Connect ready state or about to execute remote commands

murk-sy, 2023-09-25:
### Printer type
Mini/all models

### Is your feature request related to an existing problem? Please describe.
Currently, when a printer is being remotely controlled, there is no indication that such a thing is happening. While this is not an issue when a single person is operating the entire print farm, but with multiple people involved, it could easily cause printer damage or injuries, especially if some routine maintenance/basic servicing is performed without disconnecting the printer from the network.

To be clear, this is an unlikely scenario, but with open frame printers, high speeds, and high temperatures involved, some extra caution is likely justified.

### Describe the expected functionality
**When remote control is activated on the printer (Connect ready mode)**
- Display automatic start-up warning (ASW) icon in top tray, or
- Replace Prusa Mini/model name logo with Prusa Connect logo with visible ASW icon and text
- All remote commands take effect immediately

**When remote control is activated remotely** (in consideration of dangerous commands that control the heaters, fans or axes)
- Display ASW
- Play a chime (optional)
- 5 second countdown within which the remote control can be cancelled by pressing/rotating the encoder wheel
- If manual control was recently (how recent?) used and not put into ready mode, display a warning before sending dangerous commands

For reference:
ISO7010-W018 Automatic start-up symbol:
[image]
https://www.iso.org/obp/ui#iso:grs:7010:W018

To signify manual control, you can use its counterpart ISO7000-0096
[image]
https://www.iso.org/obp/ui#iso:grs:7000:0096

Quick mock-up for Ready mode:
[image]

No mock-up for tray icon - would need some extra work to make it actually visible (same icon but flashing perhaps?)

CZDanol, 2024-05-23:
Internal ticket: BFW-4938

murk-sy, 2025-12-11:
Still relevant for the long term

closed 2026-02-24 by github-actions[bot]