prusa3d/Prusa-Firmware-Buddy #3317: [BUG] M601 Pause Position Inconsistent Across Models

bdaroz, 2023-09-26:
**Printer type** - [MK4]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive** - Prusa Supplied

**Describe the bug**
MK3 series printers, on executing an M601 pause command move the extruder to the back left. MK4 moves to the back right. For prints that pause to embed items this necessitates entirely different placement between models in a print farm.

**How to reproduce**
Insert a pause on a layer in PrusaSlicer on an MK3 and MK4

**Expected behavior**
MK4 should pause to the same (approximate) position as the MK3, *or* allow the pause position (x, y, delta_z) to be configurable.

bdaroz, 2024-05-29:
Any hope on making some traction on this and gaining some consistency across models?

bdaroz, 2024-07-29:
Still an issue

BitSmith99, 2024-10-04:
When pausing the print, the MK4 bed is moved to the back of printer. If we want to adjust something, add a magnet or otherwise work with the print, it must be accessible. Pushing the bed back makes it difficult to reach parts.
PUSH THE BED FORWARD so we can work with the print.