prusa3d/Prusa-Firmware-Buddy #3322: [BUG] Mini+ BSOD precise_stepping

Sfas19, 2023-09-26:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - [5.1.0-alpha]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB drive (gcode transferred directly from prusaslicer)

**Describe the bug**
  Near the end of my print (2 hours) the printer crashes with a BSOD with the note "Precise_stepping.cp"
  The print was with clear PETG with 100% infill. A second attempt resulted in the same issue.
[image]

**How to reproduce**
  Did not print yet with another file/material. Earlier (longer and shorter) prints with the same material and settings did not have this issue. Major difference in models is that this file (with the error) had a small top area.

**Expected behavior**
  Not to crash

**G-code**
  G-code can be found in the zip file, together with the crash dump file
[precise_stepping BSOD.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12731376/precise_stepping.BSOD.zip)

**Crash dump file**
  See above mentioned zip file

**Video**
  Please attach a video. It usually helps to solve the problem.

darraghbr, 2023-09-27:
Is this a duplicate of #3288 ?

Sfas19, 2023-09-27:
Sorry, indeed looks like the same error.

closed 2023-12-17 by Sfas19