prusa3d/Prusa-Firmware-Buddy #3323: [BUG] Random crash during the start of a new print on Prusa XL 2T. Readable message is displayTask. Divide by 0.

dav991, 2023-09-26:
**Printer type** - XL with two toolheads

**Printer firmware version** - 4.7.2

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB drive (just changed to a new one because the included one got corrupted twice in a row when connected to the printer, just before starting the print)

**Describe the bug**
When starting a new print the printer crashed.
Specifics:
- I was using the second toolhead
- The crash happened while it was starting the bed leveling process. The previous print was large and it took a while to get the bed up.

**How to reproduce**
  It is not reproducible at will, the second attempt to print went fine.

**Expected behavior**
  No crash should not happen.

**G-code**
I do not think this is gcode related, but sure:
[filament holder adapter v2_0.6n_0.4mm_PLA_XL_2h38m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12731740/filament.holder.adapter.v2_0.6n_0.4mm_PLA_XL_2h38m.gcode.zip)

**Crash dump file**
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12731705/dump_buddy.bin.zip)

**Image**
[image]

**Video**
No video available.

Prusa-Support, 2023-09-30:
Thanks for reporting.

This issue was observed on Prusa MINI before, but, as per #2972 and #3278, it was most likely related to an abrupt power shortage on the micro-USB port (shut down or removed serial line-connected device).

In this case, there is only mention of the main USB-A port used for storage of print files on the USB flash drive.

Can you confirm that no other device was ever connected via a serial line (USB cable)?
Are you able to reproduce the problem by removing a USB-connected storage/device during or before certain operations?

Michele Moramarco
Prusa Research

dav991, 2023-09-30:
The only USB devices connected anywhere were the usb drives on the USB-A port (the one on the display).
My usage of the printer is to have the usb drive connected at all times and upload the files via PrusaLink from PrusaSlicer.
When I experienced the issue I remember the following operations:
1. PrusaSlicer was trowing errors when trying to upload and no files were present on the drive itself (this was checked via PrusaLink).
2. I removed the drive and checked it and it was corrupted (no partitions were on it). I reformatted the drive and connected back to the printer. It got corrupted again.
3. Switched to a known good drive that I reformatted just before connecting it to the printer
4. Trying to print a file by selecting it from the printer UI (not PrusaLink). I remember the navigation through the UI feeling sluggish. 
5. Crash
6. Save the dump and attempt to print the same file without touching anything else and it worked without additional issues.
Just before typing this reply I checked if the original Prusa drive works and none of the issue manifested, I can upload via PrusaLink, I can start printing from it. I removed and reconnected it a couple of times. I cannot reproduce the issue.

dav991, 2023-10-14:
In the last couple of weeks (while using only the original included usb drive) I have experienced a couple of crashes, always at the start of the print, after the preheating section, before bed probing.
My flow is always the same, upload the gcode from PrusaSlicer via PrusaLink and either start the printing either by checking the option in PrusaSlicer or via PrusaLink.
I will switch to another usb drive and continue monitoring; maybe the usb drive is the root cause of these crashes.

Prusa-Support, 2023-11-21:
All in all, we need the problem to be controllably reproducible to confirm a bug and possibly work on it.
However, many blue screen scenarios (BSOD) are being addressed firmware release after firmware release.

If the cause of the problem is confirmed to be the USB flash drive, please use a different one and close the issue.

Michele Moramarco
Prusa Research

dav991, 2023-11-21:
Since the alpha 5.1.0-alpha1 I have not seen the issue. I have not used the original usb drive since mentioned in this thread. I do not know which change fixed the issue, but it is obvious that I cannot reproduce it anymore.

closed 2023-11-21 by dav991