prusa3d/Prusa-Firmware-Buddy #3326: [BUG] mini homing failure y gives non-existing error page 12305

boekenwuurm, 2023-09-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
mini (original)

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.0 alpha

**Original or Custom firmware** - [Original/Custom]
original alpha software

**Optional upgrades** - [e.g. Filament Runout Sensor]
filament sensor

**USB drive or USB/Octoprint**
via USB

**Describe the bug**
After upgrading, testing belt tension and running a selftest, I run a print of homing. When trying to home the y-axis, it tries ~18 times to reach the end, fails and gives the red error screen, with error code 12305. This error code does not exist, so there are no handles on how to resolve the error.

**How to reproduce**
run homing, and if test fails, the error appears with a QR code that does not lead to a valid prusa page. (prusa.io/12305)

**Expected behavior**
a qr code that goes to a page that helps solving the problem.

**G-code**
-

**Crash dump file**
-

**Video**

[image]

murk-sy, 2023-09-27:
I believe these are generally not added or at least not public until the firmware is final in case there are some major functionality changes.

I would still leave the issue open in case it's missed after release.

i-make-robots, 2023-09-30:
I have this same issue.  Currently the only fix is to revert firmware?
https://www.instagram.com/stories/archive/18316497640096365/

murk-sy, 2023-09-30:
@i-make-robots See #3273 for fixes/workarounds, this issue is specifically about missing help pages

i-make-robots, 2023-10-01:
here was my fix: https://www.instagram.com/p/Cx08ogqPvJg/

danopernis, 2024-05-30:
@JakoobCZ Please add https://prusa.io/12305 error code for MINI

danopernis, 2024-06-04:
Fixed, thanks for reporting

closed 2024-06-04 by danopernis