prusa3d/Prusa-Firmware-Buddy #3327: Filament Sensor reads there is still filament during print, but there is not

sisuwest, 2023-09-27:
MK4

Firmware 5.0

Original

USB drive

**Describe the bug**
  The filament ran out during my print. It prompted a filament change, which I did. After removing the old filament I'm prompted to remove the filament, which I already did. There is nothing in the extruder that should be tripping the sensor. There are no other options on the screen. The only thing to do is restart the printer and lose the print. 

**How to reproduce**
  Please describe steps to reproduce the behavior.

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/65087400/9915209a-3248-4dd4-9600-5ed57f29cd05

htuter, 2023-10-04:
I came across the same problem while printing an object with 2 colors. After finishing the first color print job Prusa  Mini has asked to replace the filament albeit I have unloaded the filament it keeps asking to unload the filament. Canceling the print is the only option. This is a repeatable issue as well.

L0sWach0s, 2024-04-12:
Same on my MK4 with Firmware MK4_MK3.9_firmware_6.0.0-RC1.bbf

sisuwest, 2024-04-12:
I still get this issue. Got it earlier this week. Had to replace my
filament sensor because I was removing it so often to clear the error
during a print.

closed 2024-07-08 by github-actions[bot]