prusa3d/Prusa-Firmware-Buddy #3328: [BUG] BSOD crash when using M300 beep codes.

WhiteDragonD, 2023-09-29:
Prusa XL

4.7.2

USB

M300 gcodes are causing the printer to crash.  Been using them for years on MK2.5s.  

In the custom End Code of Prusa Slicer, 

I add the following:

;SuperMarioBrothers-OverworldTheme
M300 S0 P75
M300 S2637 P150
M300 S2637 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2637 P150
M300 S0 P150
M300 S3135 P150
M300 S0 P450
M300 S1567 P150
M300 S0 P450
M300 S2093 P150
M300 S0 P300
M300 S1567 P150
M300 S0 P300
M300 S1318 P150
M300 S0 P300
M300 S880 P150
M300 S0 P150
M300 S987 P150
M300 S0 P150
M300 S932 P150
M300 S880 P150
M300 S0 P150
M300 S1567 P150
M300 S2637 P150
M300 S3135 P150
M300 S1760 P150
M300 S0 P150
M300 S2793 P150
M300 S3135 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S2093 P150
M300 S2349 P150
M300 S987 P150
M300 S0 P600
M300 S3135 P150
M300 S2959 P150
M300 S2793 P150
M300 S2489 P150
M300 S0 P150
M300 S2637 P150
M300 S0 P150
M300 S1661 P150
M300 S880 P150
M300 S2093 P150
M300 S0 P150
M300 S880 P150
M300 S2093 P150
M300 S2349 P150
M300 S0 P300
M300 S2489 P300
M300 S0 P150
M300 S2349 P150
M300 S0 P300
M300 S2093 P300

This should play the beep tones instead if crashes w/ blue screen of death.

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12755421/dump_buddy.zip)

[CalibrationSquare_XL_50x50x02_0.6n_0.2mm_PLA_XL_2m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12755423/CalibrationSquare_XL_50x50x02_0.6n_0.2mm_PLA_XL_2m.zip)

Krumpeir, 2024-02-13:
The problem still on Prusa XL FW 5.1.2 when S>=1000 For example M300 S1567 P150. MINI and MK4 is OK.

Krumpeir, 2024-02-21:
I don't understand what you've solved, or how I'm supposed to solve it.

 

There is a PR #3686 <https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3686>  which should solve the issue.

WhiteDragonD, 2024-05-30:
comment

danopernis, 2024-07-09:
@WhiteDragonD Sorry for late reply. This bug will be hotfixed in the next patch version. Audio will be distorted and some frequencies won't play, but at least it will not crash. See https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4050 for details.

Although technically this issue was created before https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4050 let me close it as a duplicate. Thanks for understanding.

closed 2024-07-09 by danopernis