prusa3d/Prusa-Firmware-Buddy #3329: [BUG] MINI v5.1.0-alpha BSOD when displaying any gcode preview

ConteVincero, 2023-09-29:
**Printer type** - [MINI]

**Printer firmware version** - [5.1.0-alpha]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  All methods of transfer tried, USB/Prusa Link/Prusa Connect

**Describe the bug**
 The MINI will BSOD whenever it attempts to display the thumbnail preview of a print, before printing it, regardless of the method. The BSOD will appear with the title task:displayTask?

**How to reproduce**
 Whenever you:
- Send gcode to be printed over the network
- Attempt to select any gcode on the USB stick
- Insert the USB stick
 At the point at which the thumbnail preview would appear, and you confirm your desire to print, the printer will instead pause, before showing the screen, and then the watchdog will kick in, and the printer will show the BSOD 

**Expected behavior**
  A preview screen to appear, even if only briefly, before printing will start

**G-code**
  Happens with any and all gcode 

**Crash dump file**
[Dump File](https://drive.google.com/file/d/1JAuNamNbxN7tyuiTLnYUPJ-WcBRn3yrE/view?usp=drive_link)

**Video**
  [Video](https://drive.google.com/file/d/18PQfHca7UJ_slTzCUloA-ivqCRvIaw8Z/view?usp=drive_link)

ConteVincero, 2023-09-29:
Update, for funsies I tried removing the thumbnail from the gcode to see if it made a difference, but sadly it did not.

TheNkl, 2023-09-29:
is reproducible
find gcode and crashdump here:
[https://www.myqnapcloud.com/smartshare/73e9gjk44l6p7013467w4z04_ac89fk3885oq2856r37891463598g1fh](url)

TheNkl, 2023-09-30:
duplicate of
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3275

closed 2024-06-25 by github-actions[bot]