prusa3d/Prusa-Firmware-Buddy #3331: [BUG] 5.1 Alpha Mini+: Motor hammers into axis

Citrullin, 2023-09-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 5.1 Alpha

**Original or Custom firmware** - Original

**Describe the bug**
  When the bed or Hotend is at its limits and you level it, the bed or hotend hammers a little into the end and it creates a short hammering noise until it realizes its the end. This wasn't a behavior I have seen in the previous versions.

**How to reproduce**
  Move the Hotend completly to its right limit, or the hotend to its down limit. Start first layer leveling or auto homing. 

**Expected behavior**
  A short end stop noise *DUB* and not *DDDDRRR DRRR*

**Video**
  I can create one, if necessary and required. I don't want to reproduce this behavior if not needed.

marek26340, 2023-10-02:
Can confirm with my MINI too. I'm pretty sure this behavior isn't harmful to the printer in any way and it's easily preventable by manually pushing the X and Y axis closer to the middle before homing.

mtaumike, 2023-10-09:
Printer is already homing using stallguard but I don't like this noise either

MightyPork, 2023-11-03:
I now had it, multiple times, hammer so hard it decided to reboot with "obstacle on X axis".

Sometimes pulling on the belt seems to help, but tension is fine. This never happened with firmware 4.x

https://github.com/prusa3d/Prusa-Firmware-Buddy/assets/2041118/2fa90990-0cac-4f65-a028-314b3083d05d

Citrullin, 2023-11-30:
Fixed in 5.1

closed 2023-11-30 by Citrullin