prusa3d/Prusa-Firmware-Buddy #3333: [BUG] Error code: #12604 at startup even if not trying to flash the new firmware

SamuelKosmann, 2023-10-01:
**Printer type** - [MINI]

**Printer firmware version** - 5.1.0-alpha1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**The bug:**
- The error code: #12604 appears at startup even if not trying to flash the new firmware.

**How to reproduce**
- Update to version 5.1.0-alpha1
- Reboot printer

**Expected behavior**
- No error code

I suspect this happened after i set the firmware check to always in version 4.4.1 since i had trouble getting the firmware recognized by the printer. Then I upgraded to 5.1.0-alpha1. The option to change the updated firmware check from always to something different is no longer available. It now only shows the screen with the qr-code and the tutorial on how to upgrade the firmware.

fuelflo, 2023-11-02:
I'm having the same problem on 5.1.0-alpha2 with the MINI.

Only happens when USB Drive is connected.
I also formatted the drive but the error still shows up every boot, when the USB drive is connected.

Niflheim123, 2023-11-27:
same, there are also more reports on a facebook prusa mini groups https://www.facebook.com/photo?fbid=10161254518749777&set=gm.7126673067395979&idorvanity=2807333515996644

Gisi0, 2023-11-27:
yes, i think this is the same like 3522

Niflheim123, 2023-11-28:
this error occurs also with stable 5.1.0

Renba68, 2023-12-13:
hi !

Same bug for me...

Printer type - [MINI]
Printer firmware version - 5.1.0
Original or Custom firmware - Original
Optional upgrades - Filament Runout Sensor
Expected behavior :No error code

Only happens when USB Drive is connected

Before updating to 5.1.0 i had set "Update firmware" to ALWAYS...

I think the workaround is to downgrade,, modify "Update firmware" to none and update to 5.1.0

but i'm lazy and i've simply put the firmware 5.1.0 on the usb key i left as buffer, the printer is glad ... waiting for a corrective firmware...

richtw370, 2023-12-14:
Same issue here with a mini.

Only thing I can contribute is that I used by original thumbdrive and I did not erase the older firmware from the last update before adding the 5.1 file to the uSB and doing the new update.

MartinZubek, 2023-12-18:
Same thing is happening on my MINI. I need either to keep the bbf file on the flash drive or disconnect the drive during the boot. If I keep the drive inserted and the bbf file is not present on the drive I get the red screen with 12604 error.

I am on 5.1.2 which is weird as I read in the changelog that this version should address it 🤷‍♂️.

Niflheim123, 2023-12-18:
after I flashed 5.1.2 (I was on 5.1.0 initially) the CONTINUE button did not work anymore, so the only way to boot the printer was to power it ON without the usb stick, then attach the stick.

Gisi0, 2023-12-18:
> after I flashed 5.1.2 (I was on 5.1.0 initially) the CONTINUE button did not work anymore, so the only way to boot the printer was to power it ON without the usb stick, then attach the stick.

the same from my side

JackpwMurray, 2023-12-20:
this issue can be temporarily solved by leaving the .bbf file on the USB, but this is not a long term solution.

Koder22, 2023-12-24:
I found a solution to this problem on my Mini, i down graded to 4.4.1 and then changed the setting "FW UPDATE" to "off". Then i upgraded again to 5.1.0 and 5.1.2, and can boot without the firmware files.

[image]

G0MJW, 2024-01-22:
I am suffering from the same problem. Is there a solution yet, I have to boot without the drive, which is a bit of a pain now networking works.

szamelsz, 2024-02-04:
I just put an old firmware file (4.4.1) to the drive, and the error doesnt come up anymore, it wont start the firmware update process, just boots normally, in my case. Try it.

MartinZubek, 2024-02-05:
> I just put an old firmware file (4.4.1) to the drive, and the error doesnt come up anymore, it wont start the firmware update process, just boots normally, in my case. Try it.

I just tried this trick but with older 4.3.4 and it actually downgraded my MINI from 5.1.2 to this 4.3.4 version on next boot. It also did reset my printer to factory defaults along the way (I was welcomed by the initial setup screen after this involuntary downgrade).

So I would suggest against this approach (unless the version you are mentioning is actually safe as 4.3.4 is definitely not, but I won't be playing with this further as I had to setup and calibrate the printer from scratch and don't plan to risk it again).

Let's hope this mess will get resolved soon with next update.

encip, 2024-02-05:
As JackpwMurray said, just keep a copy of 5.1.2 .bbf on your drive for now. It should let you boot and not attempt to reflash.

bbolgradov, 2024-03-10:
Any progress on this issue?

Gisi0, 2024-03-11:
No, 5.12 is still the current firmware.
So far I can see.

bbolgradov, 2024-03-11:
Yes, MK4 and XL have newer firmware, hopefully we'll get an update for MINI soon with this issue fixed.

Having 5.1.2.bbf on the USB fixes it for reboots (using the reboot button) but not for full power cycles (Off-wait-On)

solty0908, 2024-03-15:
Yep, same issue here.  Always have to boot without flashdrive, then put in drive.  Makes Prusa-connect almost worthless.  I sure wish they'd find time to fix this.

G0MJW, 2024-03-15:
I don't think there is any chance of this ever being resolved. However, it seems the Bambu printer range is a much better option these days so maybe it won't be an issue for much longer.

MartinZubek, 2024-03-15:
I was able to resolve it by resetting the printer to factory defaults while being on the latest 5.1.2. I am now able to boot the printer with flash drive inserted without the need to have the BBF file present on the drive.

From what I gathered from another threads here - the original problem is related to the fact that on the newer firmwares there was removed an option about FW updates during the boot that is by default set to "Always" and on these newer FWs it's no longer possible to change the value.

Suggested solutions were to:
- factory reset the printer which will reset this option to the new default
- export the settings to file, change the value manually and then upload the changed settings to the printer
- downgrade fw to version which supported changing this setting, set it to the other option than "always" and upgrade fw again

G0MJW, 2024-03-15:
How do you export the settings to a file?

bbolgradov, 2024-03-15:
I was looking for an export option but couldn't find one. How would I do
that?

MartinZubek, 2024-03-15:
> How do you export the settings to a file?

Honestly I am not sure, I've never tried it. It was mentioned [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3609#issuecomment-1873896237). You can apparently load the config from a file (the option to do so is in the menu of the printer). I am not sure where to find the export though.

bbolgradov, 2024-03-15:
I just did "hard reset" that supposedly deletes everything and flashes again. Did all the self-tests and calibration. Still crashes on boot.

[image]

bbolgradov, 2024-03-15:
On top of that the firmware is *not* missing from the USB flash.

Prusa, please fix this!

[image]

MartinZubek, 2024-03-15:
> I just did "hard reset" that supposedly deletes everything and flashes again. Did all the self-tests and calibration. Still crashes on boot.)

That's really strange because exactly this fixed it for me few weeks ago already and I had no problems since then.

bbolgradov, 2024-03-15:
> > I just did "hard reset" that supposedly deletes everything and flashes again. Did all the self-tests and calibration. Still crashes on boot.)
> 
> That's really strange because exactly this fixed it for me few weeks ago already and I had no problems since then.

You did downgrade to 4.3.4 before the factory reset though, so I guess that's different. Might as well try that, I've got nothing to lose at this point :)

MartinZubek, 2024-03-15:
> You did downgrade to 4.3.4 before the factory reset though

That's true, I did. But just downgrade to 4.3.4 and then upgrade to 5.1.2 didn't fix it still. The factory reset on the 5.1.2 did the trick for me. Also try to remove the BBF file from the drive because I no longer have it there and it's booting fine with drive inserted.

bbolgradov, 2024-03-15:
> Also try to remove the BBF file from the drive because I no longer have it there and it's booting fine with drive inserted.

Kind of helped, thanks. Very very inconsistent behavior.

While still on 5.1.2 I removed the bbf and it booted. Then I returned it and it crashed. Then booted without USB drive and once on the home screen I inserted it, powered it off-on and it booted with the USB *with* bbf on it.

No idea what changed and how long this will last...

bbolgradov, 2024-03-15:
Ok, so here's what I tried, for completeness sake.

Flashed to 4.4.1. Update firmware setting was already "Off". Still crashed with USB+4.4.1.bbf. Changed between "Off", "On restart" and "Always" multiple times with many power cycles - it crashed on any setting with occasional boots (every 5th or so). Left it on "Off" and rebooted with a USB+5.1.2.bbf. Since it was "Off" I expected it not to prompt me to flash, but it did; I thought this option disables the prompt leaving only the alternative flash method of clicking the wheel while booting? Either that setting does nothing or I misunderstand it. Now on 5.1.2 (both before and after factory reset on it) and I'm back to square one - it occasionally boots no matter whether a bbf is on the USB drive. Tried another USB drive as well as formatting them - didn't matter.

bbolgradov, 2024-04-12:
[6.0.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC2) was released today, might be worth trying it. I doubt it's fixed there though.

franknyberg, 2024-04-12:
> [6.0.0-RC2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.0.0-RC2) was released today, might be worth trying it. I doubt it's fixed there though.

Ah, had missed that :). Think Ill hold out for the release and keep my fingers crossed. As mentioned, this issue really hampers the usefulness of Prusa link

encip, 2024-04-12:
Issue persists here with 6.0.0-RC2.

G0MJW, 2024-04-12:
Suggest raising many comments on the release candidate – perhaps someone will notice if there are enough.

 

Mike

Niflheim123, 2024-04-12:
so this issue is from octomber 1st, 7 months later no progress. It seems to me that behaviour like this is pushing people away from Prusa.

bbolgradov, 2024-04-12:
The MINI definitely feels abandoned in favor of MK4 and XL.

mmoldoveanu, 2024-04-25:
I am using the same setup with a smart outlet and its the same issue.

Prusa-Support, 2024-05-03:
Thanks for reporting.
We have been investigating this issue for some time without any chance to reproduce it.
Our developers believe this issue may partially lie on the USB flash drive but will need to collect clear evidence and reproducibility steps for further studies.

Advanced users may colone an image of their USB flash drive and send it via email to reports@prusa3d.com, together with a short explanation of the issue and a reference to this issue ([Prusa-Firmware-Buddy #3333](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3333)).

Otherwise, we would appreciate you trying multiple USB flash drives of different brands, formatted FAT32 (max 32GB) in order to possibly narrow down the problem to one or a few specific USB flash drive models, or observe the necessary steps to reproduce the problem on any flash drive.

Michele Moramarco
Prusa Research

lakyljuk, 2024-05-03:
> Thanks for reporting. We have been investigating this issue for some time without any chance to reproduce it. Our developers believe this issue may partially lie on the USB flash drive but will need to collect clear evidence and reproducibility steps for further studies.
> 
> Advanced users may colone an image of their USB flash drive and send it via email to [reports@prusa3d.com](mailto:reports@prusa3d.com), together with a short explanation of the issue and a reference to this issue ([Prusa-Firmware-Buddy #3333](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3333)).
> 
> Otherwise, we would appreciate you trying multiple USB flash drives of different brands, formatted FAT32 (max 32GB) in order to possibly narrow down the problem to one or a few specific USB flash drive models, or observe the necessary steps to reproduce the problem on any flash drive.
> 
> Michele Moramarco Prusa Research

I dont think you understand. It was mentioned in this thread already. If you had firmware version 5 with function "Automatic updated of FW, when USB is inserted during boot", it was working for version 5. But after upgrade to 6, this option is not present in menu, yet buddy board still tries to update from USB, which fails, because there is no firmware. Some people solved it by placing firmware version 5 just to get rid of it. So no, changing and formatting USB drives has no effect. It is issue that if you allowed auto update in version 5, it is now broken in version 6. So in order to get rid of  this message with version 6 installed you have to have version 5 on USB drive, which is ignored during boot, but gets rid of the error.

encip, 2024-05-03:
I think it's barking up the wrong tree to focus on the USB drive.

I see this error when plugging in all four of the FAT32-formatted USB drives I own, three of which have never held a gcode or firmware file.  I have only ever used the drive that came with the mini for FW and gcodes.

lakyljuk, 2024-05-03:
> I think it's barking up the wrong tree to focus on the USB drive.
> 
> I see this error when plugging in all four of the FAT32-formatted USB drives I own, three of which have never held a gcode or firmware file. I have only ever used the drive that came with the mini for FW and gcodes.

Yes, because this error is about issue that during flashing 5 to 6, this option to allow or disallow automatic USB upgrades was removed. So in version 6 printer always update, if it is on USB, if not, it should boot normally. But it does not, since it still remembers option from firmware 5 and is looking for file, which is not there, or is looking for some different file, cannot tell since error message is vague. This is how I understand it.

encip, 2024-05-03:
I think the issue was a setting in version 4, and this error first appeared after flashing to 5.  It has not been fixed with 6, either.

Prusa-Support, 2024-05-16:
A little clarification and an update.

There was some initial confusion on our side because if "FW Update" was intentionally set to "Always" before the firmware upgrade, then the problem should have been noticed already before upgrading the firmware - and based on an optional intended feature. Additional confusion occurred noticing that a factory reset didn't help in a few cases but now I seem to understand those edge cases where probably the user was not running the up-to-date firmware.

All in all, the factory reset should actually restore the "FW Update" to the default option for the time being, and we are working on, trying to see if alternative solutions may be possible.

Thanks for bearing with us.

Michele Moramarco
Prusa Research

FlashSystems, 2024-05-25:
I am affected by this problem on my Prusa MK4 as well. Like suggested within this bug, I've done a hard reset and reinstalled the latest firmware. The issue persisted. The printer still outputs the "Firmware file missing" error message on each power up.

To aid debugging, I've done some tests:
* The error only occurs when powering on the printer. Pressing the reset button boots without the error message.
* I've tried three different USB flash drives. The results were the same on all of them:
  * A Sandisk 16 GByte
  * The Prusa USB flash drive supplied with the printer
  * A non-name 8 GByte flash drive I had lying around
* I've tried formatting the flash drives with Windows (FAT32) and Linux (mkfs.vfat -F 32, no partitions). This makes no difference. All flash drives were empty when inserted.
* It makes no difference if the current firmware file is on the flash drive or not. The error message is shown regardless. I've only tested this with the Sandisk flash drive.

The error persisted regardless of the used drive or the operating system used to format the flash drives.

Maybe this contains some clues for what is going on.

Prusa-Support, 2024-07-12:
There is a work in progress. 🤞 
For a little insight and recommended troubleshooting steps, please check https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3629#issuecomment-2225359933.

Michele Moramarco
Prusa Research

domgiles, 2024-08-15:
Still having the issue. Like others firmware upgrades don't resolve the issue. The printer was controlled by Home Assitant and powered off remotely after use. This is no longer possible. Would love to see a resolution to this problem. I can't believe it's dragged on for so long.

G0MJW, 2024-08-15:
I don’t think there ever will be one. They know the cause but seem unwilling to address it.

danopernis, 2024-08-16:
@domgiles @G0MJW please wait for the upcoming 6.1.1 firmware. We are pretty sure this is a bootloader issue and the upcoming firmware update contains the upgraded bootloader with the fix. We were hoping to release 6.1.0 for all the printers at once, not just MK4S, but it had to be postponed in order to meet our QA standards. Bootloader updates are always tested more thoroughly as any mistake may lead to inability to boot the actual firmware. I am sorry for the delay and understand your frustration.

Niflheim123, 2024-08-27:
Hello, after the recent update 6.1.2 with the bootloader  2.3.5 the issue is fixed for me. Thanks.

boris-tk, 2024-08-27:
So far it seems to be fixed for me as well

FlashSystems, 2024-08-28:
I can also report that this issue was fixed on my MK4 with firmware 6.1.2.

Thank you for your persistence.

JackpwMurray, 2024-08-29:
Agreed, issue resloved for me as well.

CZDanol, 2024-09-02:
Allright, seems like we fixed it at last. Thank you guys for your patience and collaboration, sorry for the trouble, and see you in new bug reports! :D

closed 2024-09-02 by CZDanol