prusa3d/Prusa-Firmware-Buddy #3337: Print stopped during printing - provided  crash dump file - error code

ShadowJediJim, 2023-10-04:
Had an error mid print on my MINI+ using the Input Shaper Alpha Firmware 5.1.0

Using USB flash drive and printing with a .25 nozzle

**Crash dump file**

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12798205/dump_buddy.zip)

ShadowJediJim, 2023-10-04:
Here is an additional  crash dump file, situation occurred at a different point in the print when attempting the same item again later.
[dump_buddy2.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12807856/dump_buddy2.zip)

ShadowJediJim, 2023-10-05:
Had the error happen during a different print as well.  Screen shot of the blue screen error.  The printer freezes during the print and this blue screen error displays along with an alarm.  "precise_stepping.cp"

[image]

sekervin, 2023-10-05:
I have the same bug... 
[image]

Here's the gcode file (GitHub forced me to change it to .txt) - the error happens 35 seconds into the print - and it happens every time on my machine.  
[Precise step bug.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12821721/Precise.step.bug.txt)

ShadowJediJim, 2023-10-05:
This has now reoccurred at almost the same point in my second attempt at this print, approximately 4 hours in 71%..  

I have included the blue screen, the g-code and the error file along with pictures of how the printer stopped mid-extrusion.

[image]
[image]
[Torrga Brawnanvil_0.25n_0.05mm_PLA_MINI_5h19m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12822626/Torrga.Brawnanvil_0.25n_0.05mm_PLA_MINI_5h19m.zip)

ShadowJediJim, 2023-10-07:
Successfully printed 6 more items, then had another error - precise_stepping.cp.   This occurred at around the halfway point of a 3 hour print.  Gcode and crash file also attached.

[image]

[crash info and gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12836197/crash.info.and.gcode.zip)

darraghbr, 2023-10-08:
Is this a duplicate of #3288 ?

ShadowJediJim, 2023-10-08:
This appears to possibly be the same type of issue.  My report includes multiple additional occurrences of the problem along with the 3mf gcode and dump files.

 

Thanks you!

closed 2024-06-25 by github-actions[bot]