prusa3d/Prusa-Firmware-Buddy #3338: [BUG] <Enter comprehensive title>

JacobChrist, 2023-10-04:
**Printer type** - [MK4]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint** - [Uploading via Prusa Link]

**Describe the bug**
  Printer crashes BSOD when starting print with 6 tool print.  Note that a 5 extruder version of this file prints fine.
[image]

**How to reproduce**
  Try the attached g-code

**Expected behavior**
  Should not BOSD

**G-code**
Crashing 6 color gcode
[FlowBot 6 Color_0.4n_0.2mm_PLA_MK4IS_1h39m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12806300/FlowBot.6.Color_0.4n_0.2mm_PLA_MK4IS_1h39m.zip)

Working 5 color gcode
[FlowBot 5 Color Single Case_0.4n_0.2mm_PLA_MK4IS_1h37m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12806333/FlowBot.5.Color.Single.Case_0.4n_0.2mm_PLA_MK4IS_1h37m.zip)

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12806312/dump_buddy.zip)

tkircher, 2023-10-05:
What kind of multi-material setup are you using?

danopernis, 2024-05-30:
@JacobChrist this should have been fixed since 5.1.0 via https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/bc141586dc9044a41e6fb48b8079915b19099e93

Feel free to reopen if the issue persist.

closed 2024-05-30 by danopernis