prusa3d/Prusa-Firmware-Buddy #3339: [BUG] STEPPER MOTORS LOCKED

Manpoozle, 2023-10-04:
**Printer type** - MK4

**Printer firmware version** - 5.0

**Describe the bug**
  After any print is finished, the vertical stepper motors aren't released, and seem to stay warm indefinitely. This creates unnecessary power draw, heat, and wear.

**Expected behavior**
 The stepper motors need not be engaged, the extruder assembly stays easily in place on the lead screws while the machine is off. The stepper motors should be released after the print is finished.

rinkek, 2023-10-11:
In PrusaSlicer in printer gcode there is an "end" section.
There should me a M84 / M18 command that should turn off the steppers.
I have not checked the disabling of steppers myself, but i have found that some Marlin commands are not working.

Manpoozle, 2023-10-11:
`{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+1, max_print_height)} F720 ; Move print head up{endif}
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 X241 Y170 F3600 ; park
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+23, max_print_height)} F300 ; Move print head up{endif}
G4 ; wait
M572 S0 ; reset PA
M593 X T2 F0 ; disable IS
M593 Y T2 F0 ; disable IS
M84 X Y E ; disable motors
; max_layer_z = [max_layer_z]`

Hmm, so it should be disabling with the M84 then, no?

rinkek, 2023-10-12:
Yes, seems it should, but like i said, some other marlin commands seem not to be working

marek26340, 2023-10-13:
Hang on. Shouldn't there be a letter Z for turning off the Z steppers aswell?
Anyway. I've noticed that when a print finishes, the head moves to the parking position and starts moving up (the first time), all while it still shows the printing process on the screen as still running. Then it stops going up briefly, the print finishes on the screen and it starts going up in Z again. This could explain the Z steppers remaining active - they get reactivated after the end of a print and move Z up twice as high as the end Gcode initially did. No idea why.

rinkek, 2023-10-13:
> Hang on. Shouldn't there be a letter Z for turning off the Z steppers aswell? Anyway. I've noticed that when a print finishes, the head moves to the parking position and starts moving up (the first [...]

You're right, i missed that completely, the Z is missing. You could try and replace 
M84 X Y E ; disable motors
with

M84 ; disable motors

Manpoozle, 2023-10-14:
That’s a good idea! But how do we get it into the standard profile? I think I haven’t changed the gcode.

arekm, 2024-02-14:
Also see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3387#issuecomment-1944379538 - defaults for MK4/MK3.5 are wrong.

bkerler, 2024-02-16:
These defaults are intentional and not wrong for the MK3.5/MK4 according to the prusa devs. So this issue can be closed.

Fluffybunny96, 2024-04-05:
@bkerler Do you know the reason why this is intentional? The MK4/3.5 auto homes at every print start.
This action is understandable on other printers where they dont auto home as quick or easily as the MK4, I think that leaving the z locked will cause premature wear and it's an unnecessary waste of power.

I would be interested in what other justifications there are to keeping the Z locked when a print has finished.

danopernis, 2024-06-11:
As explained in https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3760

> This is quite an archeological task. I wasn't able to find any written sources, so I had to resort to oral tradition and I asked the elders 😄
> 
> The only answer I was able to get was that it prevents z-axis misalignment, so we do not have to go up all the way each time we start a print. That would also explain why it is only enabled for MK3.5 and MK4 (MINI only has one axis and XL starts with the plate up)
> 
> Personally I do not find that answer satisfactory (what happens on poweroff or reset?) but this is all I got.

This seems to be working as intended and is not a bug.

closed 2024-06-11 by danopernis

tkircher, 2024-08-27:
This is still an issue. Obviously if you turn the printer off the Z motors are powered down, and there isn't a full Z recalibration every time you turn the printer on. I prefer to just leave my printer on 24/7 and I think it makes sense to do that, but I don't want it burning power unnecessarily. It would be ridiculous to have to hack the MK4 so Octoprint can turn the machine on and off when this was never a problem with the MK2 or MK3.