prusa3d/Prusa-Firmware-Buddy #3341: [ENHANCEMENT] Norwegian translation for Mini, MK4 and XL

trondkla, 2023-10-05:
**Printer type** - MINI, MK4, XL

**Is your suggested improvement related to an existing problem? Please describe.**
Norwegian translation does not exists for Buddy. I've done some of the translations for MK3.

I've started translation of the [language_en.h](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/bfb0ffc745ee6546e7efdba618d0e7c0f4c909cd/lib/Marlin/Marlin/src/lcd/language/language_en.h#L25) to language_no.h and will be submitting a PR when I'm done.

Is there a readme of how I could build it to test it on my MINI, and MK4 when I get that later this year.
This issue is just to inform so not anyone else start on the same task :)  (i.e.) @pkg2000

**Describe the expected functionality**
A fully working translation of norwegian done by me.

danopernis, 2024-05-30:
@trondkla Having Norwegian translation done by community would indeed be great. I am removing the `stale-issue` label. You are requesting better documentation for the build process, we are aware that the documentation is lacking at this point. Is there anything else we can do to help you besides improving the documentation?

trondkla, 2024-05-30:
I'll take a crack at it again this weekend and add and questions I have. 😄

danopernis, 2024-08-06:
Bad bot!