prusa3d/Prusa-Firmware-Buddy #3342: [ENHANCEMENT] XL move toolhead outside print area for filament swaps

ulab, 2023-10-05:
**Printer type** - XL 5 toolhead

**Is your suggested improvement related to an existing problem? Please describe.**
I just ran out of filament on my XL which it recognized and moved to the front left. It asked me to insert new filament, purged a little and did the final "priming" extrude before continuing with the print.

Since it was done above the printed object and the nozzle is hard to reach in that position, I was not able to pull it off before the print continued. This resulted in some string/blob being squeezed into the model.

**Describe the expected functionality**
Since the XL can move outside / in front of the actual print area, it might be nice to run the nozzle all the way to the front, so it is easier to use some tweezers or similar to remove the "bugger" before the print continues.

BrienKing, 2023-10-10:
Yes, I have the same issues.  

Similar situation when you pause a print.  It just sits in mid air oozing and when you resume it doesn't do anything to make sure there is no excess filament.   

I made a video here:
https://youtu.be/CYbxdsCbAO8

Luckily it was early in the print, had it been near the end, I wouldn't have been very happy.

ulab, 2024-06-26:
It is a bummer that improvement requests are closed automatically by a bot.

danopernis, 2024-06-26:
You could have prevented closing the issue if you had commented after the `stale-issue` label was added nearly month ago.

Stale bot helps us triage issues, so we can focus on the more active issues. Nothing personal there.

ulab, 2024-06-26:
I didn't see it as personal, but the bot could at least ignore issues tagged with `enhancement` or give them a longer "stale" time.

Having to "ping" a suggestion like this every time the bot marks it as stale can quickly become tiresome.

danopernis, 2024-06-27:
I would argue that even feature requests and enhancements may become stale.

The bot won't mark the issues with assignee and we are trying to leverage that for all the new issues and issues which were marked stale but people comment on them. Ideally, we should reach a point when the bot will run out of issues, but until then, it is helping immensely.

As an exception, I am going to reopen this issue and assign it to our product department. As developers, we mostly deal with the bugs and don't really have much influence on the roadmap of new features and enhancements.

ulab, 2025-11-05:
ping.

ulab, 2026-10-05:
plonk