prusa3d/Prusa-Firmware-Buddy #3343: [FEATURE REQUEST] Set print temperature correction offset that persists through tool changes

murk-sy, 2023-10-05:
### Printer type
XL 5 tool

### Is your feature request related to an existing problem? Please describe.
Currently, if you wish to change the printing temperature during a print (for example to try and reduce stringing on a test model), the temperature will be reset as soon as the next tool change happens.

### Describe the expected functionality
When changing the printing temperature (on a specific tool), the temperature change should persist until the end of the print.
**For example:**
- First layer T: 230°C
- Print T: 235°C
- Idle T: 70°C

**After lowering the temperature by 4°C** 
- First layer T: 226°C
- Print T: 231°C
- Idle T: 66°C

Ideally, this should affect all temperatures above above a certain threshold (for example 170°C), but it's likely not too big of a deal - though avoiding underflow should be considered.

For clarity, the current printing temperature should **not** be editable, but a separate value named **Temperature offset** or similar should be used, something like this:

- Nozzle 1 Temp. Offset: -4°C
- Nozzle 1 Temperature: 226°C

Hopefully I didn't miss anything obvious, still early days with the XL for me.

n8maher, 2023-10-06:
I ran into this exact issue when trying to combat the large amount of stringing when printing with new and dry Prusament and using the Prusa-supplied slicer settings. I changed the nozzle temperature, only for it to reset after a tool change. I would also like to be able to change nozzle temperature as well as retract settings on-the-fly and have the changes persist through tool changes. This would make eliminating stringing much easier to do.

audiorazor, 2024-01-16:
Let me add my voice for wanting to see this added as well.

rfc805, 2024-01-16:
This is causing me issues right now trying to tune a silk pla profile as well.

CptDelay, 2024-04-07:
Definitely extremely annoying. And makes the option to change nozzle temps basically completely useless for multi color/material prints.

murk-sy, 2024-06-25:
Bump to keep this open