prusa3d/Prusa-Firmware-Buddy #3344: [BUG] Prusa Mini+ Crashes when submitting files through OrcaSlicer using "Upload and Print"

superslacker87, 2023-10-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 5.10A1

**Original or Custom firmware** - Custom

**Optional upgrades** - Revo

**USB drive or USB/Octoprint**
  USB/PrusaConnect

**Describe the bug**
  When attempting to send a file to the printer using the upload and print command from OrcaSlicer, the printer will crash.

**How to reproduce**
1. Slice a file in OrcaSlicer on a Prusa Mini with PrusaLink enabled.
2. Select Print and choose the option "Upload and Print"
3. The file successfully uploads to the printer
4. Instead of starting the print, it crashes out with the error in the logs attached.

This did not happen when I had the non-input shaping version of the firmware on the printer. It also is not affected if you only select Upload instead.

**Expected behavior**
  The file should upload and begin printing.

**G-code**
[flowrate_0_PLA_54m14s.gcode.log](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12836208/flowrate_0_PLA_54m14s.gcode.log)
Remove. ".log" Note that it is not gcode-related. It did this for multiple files.

**Crash dump file**
[dump_buddy.bin.log](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12836211/dump_buddy.bin.log)
Remove. ".log"

[image]

superslacker87, 2023-10-07:
Update: It is also occurring using PrusaSlicer and when using Upload or Upload and Print.

darraghbr, 2023-10-08:
Is this a duplicate of #3288 ?

superslacker87, 2023-10-11:
Does not appear to be. That appears to be an issue mid-print. Mine crashes immediately after the upload or upload and print file gets transferred to the USB drive over network.

I am able to upload the files from the web interface and upload and print also works fine from the web interface as well. But anything coming from either slicer crashes the printer.

closed 2024-06-25 by github-actions[bot]