prusa3d/Prusa-Firmware-Buddy #3345: [BUG] BSOD "precise data access" while printing

alfonsvh, 2023-10-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI+

**Printer firmware version** - 5.1.0-alpha

**Original or Custom firmware** - Original

**Optional upgrades** - [e.g. Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB drive

**Describe the bug**
  While printing, BSOD appeared stating "precise data access".

**How to reproduce**
   Unable to consistently reproduce.

**Expected behavior**
  Nothing 

**G-code**
[Mayan_Death_Whistle_0.4n_0.2mm_PLA_MINIIS_2h16m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12837227/Mayan_Death_Whistle_0.4n_0.2mm_PLA_MINIIS_2h16m.zip)

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12837226/dump_buddy.zip)

darraghbr, 2023-10-08:
Is this a duplicate of #3288 ?

alfonsvh, 2023-10-08:
> Is this a duplicate of #3288 ?

Don't think so, different error

darraghbr, 2023-10-08:
Ah apologies I had the wrong issue in mind is it the same as #3311 ? I also think the error is too similar to not be related to #3304 Surely 'Imprecise Data Access' and 'Precise Data Access' are at least linked...

alfonsvh, 2023-10-10:
> Ah apologies I had the wrong issue in mind is it the same as #3311 ? I also think the error is too similar to not be related to #3304 Surely 'Imprecise Data Access' and 'Precise Data Access' are at least linked...

The 3311 does indeed seem very similar; although the hex dump differs it seems to be the same error indeed.

closed 2024-06-25 by github-actions[bot]