prusa3d/Prusa-Firmware-Buddy #3346: Nozzle Cleaning Step Fails on Nearly All Prints Using PETG

IZ-Labs, 2023-10-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK3.9]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - []

**USB drive or USB/Octoprint**
  USB Drive

**Describe the bug**:
Nozzle cleaning fails on nearly every print when using PETG. The printer tries two 'rows' of cleaning steps and the shows dialog saying "Nozzle cleaning failed". Visual inspection of the nozzle shows little if any filament, and if any is present at all, it is something that would be cleaned during the pressure equalization wipe. I'm using brand new Prusament PETG, sliced on default settings in Prusa Slicer

I imagine this is happening as the preheat temperature during this step is only 170C, which doesn't allow PETG to melt and actually get cleaned off the nozzle as PLA might at these temperatures. Perhaps an update to the slicer config to change the preheat temp depending on filament would fix this?

**How to reproduce**:
 Start any print using PETG. (sliced with PrusaSlicer 2.6.1+MacOS-arm64, Prusament PETG profile)

**Expected behavior**:
Nozzle cleaning should execute and remove any debris from the nozzle in a couple seconds, then allow the print to continue.

**G-code**:
  See attached
[NozzleFailExample_0.4n_0.2mm_PETG_MK3.9_26m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12841815/NozzleFailExample_0.4n_0.2mm_PETG_MK3.9_26m.gcode.zip)

**Crash dump file**:
  N/A

**Video**:
  Will try to add this soon.

m-cas, 2023-10-09:
I have an MK4 and get the same error with PCBlend about 60% of the times. No issues with PLA and PETG. I am using a Prusa enclosure. There are a few other reports similar to these in the forum.

jpaxer, 2023-10-09:
This is happening for me too, also with PETG.

tkircher, 2023-10-10:
What firmware version did this start happening?

IZ-Labs, 2023-10-10:
I only just got my upgrade completed, so 5.0.0, but looking on here I see people reporting this issue dating back to the 4.x.x releases

loewi1979, 2023-10-11:
i have the same issue on PLA and PETG (new MK4 with Firmware 5.0.0)

tjhawkey, 2023-10-12:
The new configuration bundle changes the PETG temp to 195 from 170. I haven't checked other filament types

seakrebel, 2023-10-13:
Same here. MK4 with newest firmware.

incof, 2023-10-19:
I have the same problem. Sometimes help if I low tension force of filament from filament spool. It seems that tension sensor is reacting to it. PLA ang PETG - the same.

m-cas, 2023-10-19:
I have a very low friction mount for the spool and I use it for all my prints... I really wonder why some of us has the problem with PETG and other only with PC Blend...

codingcatgirl, 2023-10-23:
Experiences this as well now, with Prusament PETG. For a while it was just failing once, i would retry, maybe clean the nozzle a bit, and it would work… but if felt like cleaning the nozzle was usually not necessary.

But now i'm on the fourth attempt and it just can't get past it. No clog.

TravisWilder, 2023-10-23:
5.1a2 with PS2.7a on MK4 has PETG cleaning with 195°C which is for me way to hot and caused issues.
I changed the start code to 180°C which worked much better

m-cas, 2023-10-23:
> 5.1a2 with PS2.7a on MK4 has PETG cleaning with 195°C which is for me way to hot and caused issues. I changed the start code to 180°C which worked much better

Any chance you can publish here your changes?

TravisWilder, 2023-10-23:
> > 5.1a2 with PS2.7a on MK4 has PETG cleaning with 195°C which is for me way to hot and caused issues. I changed the start code to 180°C which worked much better
> 
> Any chance you can publish here your changes?

These two lines I change the 195 to 180

M104 T0 S{((filament_type[0] == "PC" or filament_type[0] == "PA") ? (first_layer_temperature[0] - 25) : (filament_type[0] == "FLEX" ? 210 : (filament_type[0]=~/.*PET.*/ ? 180 : 170)))} ; set extruder temp for bed leveling
M109 T0 R{((filament_type[0] == "PC" or filament_type[0] == "PA") ? (first_layer_temperature[0] - 25) : (filament_type[0] == "FLEX" ? 210 : (filament_type[0]=~/.*PET.*/ ? 180 : 170)))} ; wait for temp

SmithyAT, 2023-10-23:
Same here with Prusament PETG and MK4 FW 5
Will try the workaround to lower the temp.

RedKrieg, 2023-11-09:
I've been able to work around this by reducing preheat temp to 180C as demonstrated by @TravisWilder but I still have to be certain my nozzle is spotless before the print starts, which has never been a problem with PLA.  I have to watch the first layer very carefully and if I spot any gaps I have to stop the print and start over.  Dropping preheat from 195->180 has changed my success rate printing PETG from 20% to closer to 70%.

rsheldiii, 2023-12-24:
This is also happening to me. Is there any ability to turn the cleaning check off?

Gottox, 2024-01-06:
Same for me here.

folays, 2024-04-12:
This also happened to me on a MK4, on a completely new-empty-clean nozzle.

I remplaced the 0.4 brass nozle with a NEW-empty-clean ObXidian 0.4 ("Nextruder" version),
and then I tried to print my 1st `Prusament PC Blend Urban Grey` on a clean smooth PEI sheet.

Note that I did *NOT* put glue stick on the clean/purge left-front area.
Normally I put glue stick (as recommended) on the print area, to print PC Blend on  smooth PEI sheet.
It just didn't occurs to me that maybe I should put some on the cleaning area also.

The printer printed two rows of dots on the left-front sheet, and then "Nozzle cleaning failed".

Then I cleaned everything (which was hard, since there were no glue stick), and it succeeded the 2nd time.

So it's weird that it failed to "clean" what was an empty nozzle...

I've never had this error before (with PLA or PETG).

closed 2024-07-04 by github-actions[bot]