prusa3d/Prusa-Firmware-Buddy #3347: [BUG] Nozzle cleaning fails when using PETG MK4

jpaxer, 2023-10-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4
**Printer firmware version** - 5.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint**
  USB Drive

**Describe the bug**
  Some of the time when using PETG (Default Prusament PETG settings in Prusaslicer), it will say "Nozzle Cleaning", slowly probe the bed about 10 times and then say "Nozzle Cleaning Failed" with an option to abort or retry. This happens about 50% of the time.

**How to reproduce**
  Try printing with Prusament PETG (I haven't tested with other filaments) and it will probably eventually happen.

**Expected behavior**
  I expected it to successfully finish the nozzle cleaning and continue with the print.

**Crash dump file**
  I don't know how to do that

darraghbr, 2023-10-10:
Is this a duplicate of #3346 ?

jpaxer, 2023-10-10:
> Is this a duplicate of #3346 ?

Yes, it is. Just making this to try to show that a lot of people are experiencing this issue.

tkircher, 2023-10-11:
I have no problems at all with PETG, PLA, HTPLA, metal PLA, nylon, or TPU on my new MK4. Used PETG with the stock 0.4mm nozzle, used the nozzle adapter + diamondback for abrasive PLA. Just because you are having a problem doesn't mean it's a problem with the printer.

Blanchegorge, 2023-10-13:
Sale problem for me. With prusa PETG

seakrebel, 2023-10-13:
Same problem with my MK4, Prusa PETG as well. 
Didn't have this problem in the first two months of usage, but now I have it on every print.

Blanchegorge, 2023-10-13:
After managing to clean the head, I launched an impression that failed, the head was clogged. I then changed the petg by asa to unclog (higher temperature) and the prints worked well in asa. Then I put the petg back and it worked (it was another color). The problem was with the black petg.

SmithyAT, 2023-10-23:
Same problem here every 2nd print with orange Prusament PETG. 
Never got that error with PLA.

pgroll666, 2023-10-25:
Same Problem for me on the new MK4
Printing PETG on textured sheet failes every second try due to "nozzle cleaning problem". Very annoying as the print does not continue. Reset sometimes helps.
PLA on smooth sheet is ok.
What is the reason for this ? is it meant to verify wether the nozzle is clean or should it clean the nozzle ?

SmithyAT, 2023-10-25:
It is interesting, that I have now changed the temp for pets to 180, and never seen a nozzle cleanup step anymore. I just run through the start process, leveling and so on.
Or is the nozzle cleanup not shown on the display, just when an error occurs? 
But my workaround for now is to change the temp in the start gcode.

Lily2012AK, 2023-10-26:
I had no issues since I received my MK4 sometime in June, until last night.  The only difference I noticed was the higher temperature for bed leveling, which increased from the usual 170c to roughly 190c.  As @SmithyAT mentioned the PETG seems to be oozing but not quite hot enough to stick to the bed and clean the nozzle so the filament just piles up and with every pass the nozzle goes higher.  I believe that is the reason for failed nozzle cleaning, at least in my case.  Any info on why the temperature changed?  I don't remember updating any configurations and printed successfully last night with PETG.  170c is working fine as long you clean the nozzle before it starts the print.

tkircher, 2023-10-26:
Something to consider is that, if the problem is everyone printing their PETG too hot, that's not a firmware issue at all. That would be an issue with the default slicer profiles.

Crestburg, 2023-11-16:
I have the same issue and it just started yesterday out of nowhere. Factory assembled MK4 in a Prusa enclosure running PETG that is the second spool of the first spool that run out but has printed a bunch of prints. Happens almost every time but after I press the reset it seems to work. Very annoying as it just seem to happen for no good reason. Bed is clean and I use the Prusaslicer with the Generic PETG settings.

TillBerthold, 2023-12-15:
Is there still no solution for that problem? It seems like Prusa is not interested in fixing it. My Mk4 is almost unusable because of that problem.

seakrebel, 2023-12-15:
> Is there still no solution for that problem? It seems like Prusa is not interested in fixing it. My Mk4 is almost unusable because of that problem.

I reduced the extruder temperature for bed leveling to 170 degrees, and I haven't seen this error for a while now. 
You can change this by switching to **Expert** mode in PrusaSlicer, and then going to **Printer Settings > Custom G-code** and adjusting the code around line 14 to look as follows:

```
M104 T0 S{((filament_type[0] == "PC" or filament_type[0] == "PA") ? (first_layer_temperature[0] - 25) : (filament_type[0] == "FLEX" ? 210 : (filament_type[0]=~/.*PET.*/ ? 195 : 170)))} ; set extruder temp for bed leveling
M109 T0 R{((filament_type[0] == "PC" or filament_type[0] == "PA") ? (first_layer_temperature[0] - 25) : (filament_type[0] == "FLEX" ? 210 : (filament_type[0]=~/.*PET.*/ ? 195 : 170)))} ; wait for temp
```

TillBerthold, 2023-12-15:
Thanks for your answer. The code already looks like that.
I will check again what the temperatures while at nozzle cleaning.

closed 2024-06-25 by github-actions[bot]