prusa3d/Prusa-Firmware-Buddy #3350: [BUGS] <Very bad firmware with a lot of bugs !!!>

catagro, 2023-10-10:
**Printer type** - [MINI - clone]

**Printer firmware version** - [e.g. 4.0.5, ...]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Direct drive]

**USB drive or USB/Octoprint**
  - USB drive

**Describe the bug**
  - The presented benchy on the new software page **was made with an MK4 printer !!!!!**
  - The firmware does not work good with other printers (it is normal ...)
  - ****You have eliminated** from the firmware **the FYSETC display type** from the new firmware and all who have clones their display are flashing !**
  - Together with the new firmware you are giving a benchy wich was sliced with  **Prusa Slicer Version 2 6 0 - Beta** 2
  (From where did he take-it ? **Do you really think we are all idiots ?** Justa asking, no offence !!! )
  - The total and remaining time are not displayed !
  - After PAUSE the display is not remaining on RESUME but it is on the STOP icon !
  - After the print is finish the **motors are NOT DISABLED** !!! 
  - etc, etc, etc ...
  - **Are you sure you are Prusa Research ?**

**How to reproduce**
  - Just slice yourself with the firmware a NORMAL benchy and it will be a mess like the one presented which we have try to print !

**Expected behavior**
  I expected to be a better software for better printing but is a mess !

**G-code**
**The benchy Gcode attached on the firmware page was made with an MK4 printer and sliced with Version 2 6 0 - Beta 2" SLICER**
It's not normal - not all of us we have an MK4, you must make a firmware and a slicer for THE MAJORITY of us !!!

  **M862.3 P "MK4" ; printer model check**
  M862.5 P2 ; g-code level check
  M862.6 P"Input shaper" ; firmware feature check

**Crash dump file**
  - No dump file ~

**Video**
  **I am curious from WHERE HE HAD TAKE THE "Version 2 6 0 - Beta 2" of the Prusa SLICER ????**
  .
[image]
*

Jerzeek, 2023-10-10:
Dear @catagro 
I find the issue you opened here not very constructive. I am a very happy Prusa Mini user and I am amazed at all the support and updates the Mini has gotten so far. 
If you are not satisfied with the new firmware. Please keep using the old firmware or propose changes yourself.

catagro, 2023-10-10:
> Dear @catagro I find the issue you opened here not very constructive. I am a very happy Prusa Mini user and I am amazed at all the support and updates the Mini has gotten so far. If you are not satisfied with the new firmware. Please keep using the old firmware or propose changes yourself.

Dear @Jerzeek  "a very happy Prusa Mini user" !
Thanks God you gave me the permission to reflash the old firmware, other way I would be lost :)))))))))))
About my "Issue" which is not "very constructive" ...
What proposition from all the mentioned bugs did you not understand ?
It is real what you said that Prusa have made a "lot of updates" I can't deny.
About support, if I am looking to all the complaints I am not so sure ...
Anyway, I want to tell you I have ... many years like user of the 3D printers and even I'm not a software engineer (I'm a mechanical one) before Prusa I have compiled more than 1000 times Marlin and klipper from different configurations so I know what I'm talking about, is it clear for you ?
About a single bug like example: to launch a firmware which don't stop the motors after the print it's ... at least a shame for a company like Prusa !
I don't have the time to study the gived code  in the Github from Prusa, code which is a combination between Marlin (which is very clear and logical), Klipper (also), and many, very many other scripts which are calling functions God knows from where if you don't study at least a week the code, Maybe I will, I will see, but that case I will ask Prusa big a reward for my work !!!
About me, I will tell you a single fact: 
**I am "THE ONE" when the last firmware was launched and nobody could flash-it who have find the solution to change the name (I really don't remember if for the bootloader or the firmware) !!**
After that all have succed to flash, after myGithub  post  !!!
How do you think I've done-it ?
I guessed, I was to a sorcerer or what ? 
No, I've opened the code and I saw the error !
Please receive all my best @Jerzeek but mind your own business (if you are no a Prusa team member, which I am believing for real)

beelsebob, 2023-10-10:
I’m in love with the fact that you think it’s Prusa’s job to support your clone printer with non-standard hardware in it.  Like, they’re going to psychically know what is going to be in all the random things people might install their firmware on.  Here’s a hint though: if you bought your printer from fystec, it’s fystec’s job to provide firmware for it, not Prusa’s.  You can’t just install prusa firmware and expect that they’ll support a competitor’s hardware.

catagro, 2023-10-10:
> I’m in love with the fact that you think it’s Prusa’s job to support your clone printer with non-standard hardware in it. Like, they’re going to psychically know what is going to be in all the [...]

@beelsebob I will not argue anymore, do all of you as you pleased. 
Personally I have succeeded to recompile the firmware and eliminate all the bugs so I'm not interested anymore from a Prusa help !
Good luck to all of you !

matrndev, 2023-10-11:
So this issue was opened just because the OP wanted to tell the world that he recompiled the firmware for _his use_.
That's not how open-source works.

catagro, 2023-10-11:
> So this issue was opened just because the OP wanted to tell the world that he recompiled the firmware for _his use_. That's not how open-source works.

@matrndev I think you don't understand !
1. I am not a member of the Prusa team, I am a private person and I have a Prusa Mini clone !
2. Open source is what Pusa have posted on Github, anyone can take the source and to modify-it how he/she wish but THAT DON'T MEANS HE HAVE THE OBLIGATION TO MAKE PUBLIC HIS WORK  !!!
3. This Issue was open in the idea that Prusa will do something, instead I am being attacked (see the first two comments)
Because I was "attacked" I had renounce to wait an answer from Prusa developers and I have made my own modifications.
IS IT CLEAR FOR YOU ?

Lino77, 2023-10-12:
[mini_release_boot_5.1.0.bbf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12881885/mini_release_boot_5.1.0.bbf.zip)
the display has not been removed. it's just incompatible. but a firmware especially for you that runs with your Fystec display,

catagro, 2023-10-12:
> @catagro https://youtu.be/OdsH2qSfYVo Oh, I just see that they have already fixed it themselves. then be so kind and close this commit

@Lino77 Thank you for the file !
I will close the commit, anyway I wanted to do this.

Lino77, 2023-10-12:
@catagro die wurde nicht veröffentlich die habe ich selbst erstellt

catagro, 2023-10-12:
> @catagro die wurde nicht veröffentlich die habe ich selbst erstellt

@Lino77 Ich verstehe. Vielen Dank