prusa3d/Prusa-Firmware-Buddy #3351: [BUG] M303 (PID autotune) Unknown command

rinkek, 2023-10-10:
When starting the PID autotune, M303 E0 S210 C10, it returns: Unknown command: "M303 E0 S210 C10"

earglass, 2023-12-14:
Same here. Firmware 5.1.0 - Prusa Mini+
Was PID tuning somehow removed from this version?

kozross, 2023-12-15:
@rinkek and @earglass - does this work on 5.2.0?

bkerler, 2023-12-21:
Not sure what you mean with 5.2.0, I assume you mean 5.1.2 ?

kozross, 2023-12-21:
@bkerler - I did. Will edit my response.

bkerler, 2023-12-21:
I had a look at the source code and it's disabled (PID_AUTOTUNE is not defined) for normal configurations (regular firmware). There is an advanced configuration where it is enabled (I assume for developer tests). I will ask if there is a special reason why it is currently disabled.

bkerler, 2023-12-22:
Maybe for a better understanding here why some features from 4.xx are missing in 5.xx : Those features weren't removed as many guess but instead the firmware was rewritten from scratch and some features weren't added yet because they do not make any sense due to newer changes or simply because they haven't yet been reimplemented. The PID autotune is still implemented in the marlin gcode base (which isn't used anymore in the prusa fw 5.x), but not yet in the new gcode base.

bkerler, 2023-12-22:
Also bear in mind the flash space on the mini is limited. So there may not be enough space for both input shaping and other stuff. So in the end you have to decide if you want input shaping with missing stuff or no input shaping with all other stuff you're missing now.

bkerler, 2023-12-22:
Reviewing the firmware notes [here](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3), it seems that pid doesn't work correctly on the mini due to its hardware. I assume you have a very good reason why you still need the pid autotune @rinkek. Can you please provide data or more detail why it is needed ?

espr14, 2023-12-22:
MINI has not enough flash and RAM. So they removed:
- all but 2 languages
- PNG decoder (replaced by QOI)
- unecessary stuff

Let's hope they will:
- optimize code like in MK3
- release MINI MK2 with better MCU

bkerler, 2023-12-22:
Yes, I truely think at least offering a board with more ram + flash would be really necessary for the mini. Doesn't need to be mk2.

closed 2024-06-25 by github-actions[bot]