prusa3d/Prusa-Firmware-Buddy #3352: [BUG] Power flicker turns off heat

3dGeneDesigns, 2023-10-10:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** - USB

**Describe the bug**

My power flickered just fast enough to not trigger power failure mode but still turned off the nozzle and heat bed temp. Came back to check on it and saw it was still doing the movements of printing but had a major gap between top layer and nozzle. Time frame matches up with the power flicker my house had.

**How to reproduce**
  Don't think I can

**Expected behavior**
Full power fail mode for any major voltage change.

**G-code**

[Gnome x2_0.4n_0.3mm_PLA_MK4IS_3h53m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12862352/Gnome.x2_0.4n_0.3mm_PLA_MK4IS_3h53m.zip)

tkircher, 2023-10-11:
It should at least fail fully or not at all, rather than failing halfway, for sure.

3dGeneDesigns, 2023-10-11:
Figured out it was actually a gcode fault.

closed 2023-10-11 by 3dGeneDesigns