prusa3d/Prusa-Firmware-Buddy #3354: [FEATURE REQUEST] Implement some kind of USB camera support on the xBuddy platform

Zappes, 2023-10-11:
**Printer type** - Mk4 (Firmware 5.0.0)
**Optional upgrades** - None

**Is your feature request related to an existing problem? Please describe.**
It would be very helpful for remote printing if one could attach a USB web cam to the printer and monitor the progress from the PrusaLink UI. Such a functionality exists in the version for the RPI, but not for the embedded PrusaLink in the Mk4

**Describe the expected functionality**
When a camera is attached to the USB C port, it should be detected. The image captured by the camera should be displayed on the PrusaLink UI.

Given the limited power of the controller, full video stream support is probably not an option, even though this would be the best solution. Other options would still be really useful, though:

- Capture a screenshot every n seconds.
- Capture a screenshot on layer change

BernhardFe, 2023-10-12:
I have bought, because of this issue, the beagle cam 2 for my MK4. In former times I used Octo Print with a camera for the MK3. 
The beagle Cam is running independently on its own but gives some options to stream the print situation to my mobile. Maybe this can be connected to the buddy platform to fullfill the upper use case? The connection might be done via W-Lan for the buddy platform direct and as well for the Prusa Connect. I found no way to do this. So all is running independent at my printer.

Zappes, 2023-10-18:
I also bought the Beagle v2 in the meantime, but that's also a semi-satisfactory solution as the Mk4 withholds a lot of information from the device - the temperatures, for example, are not shown in the Beagle UI. :( I guess one can't pressure Prusa to unfuck the connectivity of a third-party device, so the request to Prusa basically stays "if you don't want to support external solutions, please do at least add camera support like they have it".

CommanderRedYT, 2023-10-23:
The question is if this is possible from a hardware perspective. Where would you plug in such a camera?

Zappes, 2023-10-23:
The USB C ports seems to be an obvious choice.

CommanderRedYT, 2023-10-23:
> The USB C ports seems to be an obvious choice.

Doesn't have to. It depends on how it is wired. It might only have some kind of serial connection like with an Arduino, then this wouldn't be possible. It has to be a port capable of being a USB Host Interface.

BernhardFe, 2023-10-23:
Beagle Cam could be connected via Wifi commands and commands can be send from prusa connect or Prusa Link/Prusa connect. There is an option to steer the printer via Beagle interface on the mobile with the beagle app. But the other way around might be also hopefully possible. I mean that the video would be available in Prusa Connect Interface

BernhardFe, 2023-10-23:
Yes, but an easy way would be to connect the Beagle Cam to the prusa connect like you can do it with a smart phone. But I need my smart phone... :-) So to embed the stream of the beagle cam would be great.
[image]

cc2224cody, 2024-06-05:
Still a wanted feature! Please do not close.

Scrazzled, 2024-06-06:
I'd also like this feature as well.