prusa3d/Prusa-Firmware-Buddy #3356: [BUG] BSOD on startup

Skaronator, 2023-10-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL5Tool

**Printer firmware version** - 4.7.2
**Original or Custom firmware** - Original
**Optional upgrades** - Normal XL 5 Tools

**USB drive or USB/Octoprint** USB Drive and Prusa Connect

**Describe the bug**

I turned the power on and the printer BSOD without any user input. I also use PrusaLink (Home Assistant will fetch some status) and Prusa Connect. I suspect that one of these two services might be the issue.

[image]

**How to reproduce**
Not sure, this was the first time.

**Expected behavior**
No BSOD

**G-code**
N/A

**Crash dump file**
[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12878310/dump_buddy.zip)

Skaronator, 2023-10-20:
Thanks @dav991! I can confirm that is basically how it went for me as well.

I've uploaded gcode to PrusaLink but without starting the print, waited like 10 minutes to make sure the file didn't get corrupted (looking at my MK3S with PrusaLink), turned the printer off with my smart plug. The next morning, I turned the printer on and checked like 20 minutes later to find the BSOD.

Scary that the printer actually starts a print - this could have ended horribly!

dav991, 2023-10-22:
On firmware [5.1.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2) I cannot reproduce the issue anymore.

Skaronator, 2023-10-22:
Just did a quick test and seems to be fixed. Thanks for verifying as well @dav991.

closed 2023-10-22 by Skaronator