prusa3d/Prusa-Firmware-Buddy #3358: [FEATURE REQUEST] Painting the BRIM position like the support positions can be painted

BernhardFe, 2023-10-12:
Situation in Prusa Slicer today: 
When I need to print with brim I have sometimes the problem that the brim should be not at special places of the print part. It can be difficult sometimes to take it away without damage the print part. 
Proposal for a new function: 
Being able to draw the brim in the same way like the support can be painted and setup manually. 
Then I can exclude critical areas or can reduce the brim to a minimum. 
This would be great for the part and less work to take the BRIM away.
BR Bernhard

mix579, 2023-10-12:
This should be moved from Firmware-Buddy to Prusaslicer. it's also a duplicate of #6208 and #8283 (and probably more) in that repo.

BernhardFe, 2023-10-13:
Oh sorry, you are right... I will merge it with #8283.

closed 2023-10-13 by BernhardFe