prusa3d/Prusa-Firmware-Buddy #3361: [BUG] BSOD: fsm_types. cpp: 361

meermanr, 2023-10-13:
(I have searched for this bug, and not found it)

**Printer type** - MINI

**Printer firmware version** - 5.1.0-alpha1+12820

**Original or Custom firmware** - Original

**Optional upgrades** - Filament runout sensor, E3D Revo, 0.6mm nozzle (brass)

**USB drive or USB/Octoprint** - USB

**Describe the bug**
Blue screen of death while uploading g-code via PrusaLink from PrusaSlicer:

```
BSOD
fsm_types.срр:361
FSM DESTROY: PrintPreview does
not exist. Called from void
marlin_server::_server_print_l
oo?р
(),/home/admin/workspace/RELEA

5.2.0-alpha     [A]
```

**How to reproduce**
(Not able to reproduce)

**Expected behavior**
Upload completes, g-code stored on USB Mass Storage, print begins.

**G-code**
[Spool holder - one sided triangle_0.6n_0.3mm_PLA_MINI_2h56m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12899786/Spool.holder.-.one.sided.triangle_0.6n_0.3mm_PLA_MINI_2h56m.gcode.zip)

**Crash dump file**
[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/12899787/dump_buddy.bin.zip)

**Video**
Photo of Blue Screen of Death:

[image]

meermanr, 2023-10-13:
Note that copying the same G-Code file to the USB drive, inserting it into the MINI, and then loading it via the usual on-screen menus works, but lacks a preview picture.

darraghbr, 2023-10-15:
Is this a duplicate of #3280 ?

meermanr, 2023-10-15:
Yes, it seems to be, but I didn't find it because it lacked the actual error badge as searchable text.

danopernis, 2024-06-03:
This particular type of BSOD should be permanently fixed since v6.0.0, fixed via https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/99da8a4f253df34dc34f5f81f3607b94498ae96e

closed 2024-06-03 by danopernis