prusa3d/Prusa-Firmware-Buddy #3370: [BUG] cannot swap the filament after a filament runout event and cooldown happens

metebalci, 2023-10-18:
**Printer type** - [MK3.9]

**Printer firmware version** - [5.0.0]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [None]

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
When filament is runout yesterday, I did not hear the beep and the printer went idle (the hotend cooled down). When I realized this ~an hour later, I saw the runout filament piece is unloaded and it was saying change the filament etc. I removed it, and tried to load a new one. However, it was always saying remove the filament, and looked like it did not detect I removed the old one and trying to load the new one. The filament sensor does not have a problem, I am using it all the time, never had an issue. Inserting (pushing inside) or removing (pulling out) the filament did not change anything, it looked like either the filament sensor was not working (which is not correct electrically) or the software was stucked at something and could not process the information from the sensor.

**How to reproduce**
I did not reproduce it but a filament runout could be triggered, and then wait for printer to go idle, hotend to cool down. Then, try to load a new filament.

**Expected behavior**
It should be possible to swap the filament after a filament runout event.

**G-code**
I did not keep the g-code file but it was a g-code produced by PrusaSlicer (2.6.1) from a model published in printables, nothing unusal.

**Crash dump file**
None

**Video**
I did not take a video unfortunately.

RichiH, 2023-10-23:
Confirming the same issue for 5.1.0-alpha2 on the Prusa XL.

RichiH, 2023-10-23:
Addendum: It started heating the nozzle the second I pressed the "not idle any more" dialogue, but still tried to unload the filament while cooled off.

Sdee619, 2023-10-28:
Same issue with the filament runout sensor not triggering when doing a filament swap

closed 2024-06-26 by github-actions[bot]