prusa3d/Prusa-Firmware-Buddy #3373: BSOD: XL 2-tool, cancel current item with single-item print

ex-nerd, 2023-10-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - XL, 2-tool

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint** - USB + wifi

**Describe the bug**

Used the new "cancel current object" feature and it cancelled but did a bunch of weird X/Y movements as it was lowering the Z away from the print. Then BSOD.

**How to reproduce**

I just used "cancel object ... current object" in the middle of a print of a single object.

**Expected behavior**

Print should have been cancelled without a BSOD.

**G-code**

Sorry, the file I was printing is licensed such that I cannot provide this.

**Crash dump file**

[dump_buddy.bin.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13057998/dump_buddy.bin.zip)

**Video**

N/A

816-80085, 2023-10-23:
I had exact same bug. Same hw, FW, same events.
[image]

ex-nerd, 2023-10-23:
I missed a screenshot when I filed this. I just noticed from yours that this is the same line of code as the other bug I filed at #3386

closed 2024-06-26 by github-actions[bot]