prusa3d/Prusa-Firmware-Buddy #3374: [BUG] <Enter comprehensive title>

Tommy544, 2023-10-20:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]
MINI+

**Printer firmware version** - [e.g. 4.0.5, ...]
5.1.0-alpha2

**Original or Custom firmware** - [Original/Custom]
Original

**Optional upgrades** - [e.g. Filament Runout Sensor]
Filament Runout Sensor

**Describe the bug**
The option to display currently selected Steel Sheet profile in the Footer disappeared. It was present in alpha1 and other previous firmware versions. Instead, it looks to me like there are two Z axis coordinate entries in the list. They show the same value, but one rounds it to one decimal place and the other one to two decimal places. (There is a third entry in the list with the same Z logo, but I guess that's a different metric)

**How to reproduce**
Go into the User Interface -> Footer menu and try to set any of the 3 configurable positions to show currently selected Steel Sheet profile.

**Expected behavior**
The option to display currently selected Steel Sheet profile is important, as you can easily damage your sheet if you accidentally didn't select the right Steel Sheet profile. Having it displayed in the Footer at all times is a great help.

Stauc, 2023-10-21:
The third Z is the live Z setting, I use that now to determine the selected Steel Sheet, 
but please bring back the Steel Sheet in footer, we sometimes forget to change the steel sheet, or the setting when changing filaments and need that reminder :)

MankeyJohn, 2023-10-23:
Thanks for the work around, Will have to keep that in mind. Such a useful Footer!

gegy1, 2023-10-28:
This was realy a very handy feature on the mini for last check if you have the right steel sheet on your bed, before starting the print. Even if the print has already started, you can check and stop if wrong sheet is set. Please bring back this footer info.

SnowB0und14, 2024-02-22:
I just noticed this same issue on the MK3.5.  It is super obvious when transitioning from the MK3s+ to the MK3.5 because it is missing from the status screen and you cannot add it to the footer.  Seems like an easy fix.  I just tore up my PEI sheet because I didn't realize it wasn't selected.

cakesmith, 2024-04-12:
+1 I would also like the sheet info on my footer, please.

CZDanol, 2024-05-16:
Closing as a duplicate of #3789 . This one is older, sorry, I screwed up a bit :D

closed 2024-05-16 by CZDanol