prusa3d/Prusa-Firmware-Buddy #3375: [BUG] Alpha2 MINI+ BSOD error

osvch, 2023-10-21:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
 USB flash drive

**Describe the bug**
 I sliced benchy with PrusaSlicer 5.1.0-alpha2, copied to USB drive. After selecting model and hitting print I receive fault as attached heap.cpp:37
Attached zip is buddy dump and binnary gcode.

**How to reproduce**
Slice new model, export to usb drive, select print- fault occures

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.
[image]
[dump_buddy_gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13060617/dump_buddy_gcode.zip)

osvch, 2023-10-21:
UPDATE:
I've done hard reset of printer and issue no more present.

abstractFlo, 2023-10-22:
What do you mean with hard reset? I have this error after start a new print if one was done before

Pitel, 2023-10-22:
Same crash, but when sendig from Slicer over Wifi.

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13064188/dump_buddy.zip)

osvch, 2023-10-23:
> What do you mean with hard reset? I have this error after start a new print if one was done before

Factory reset -> Hard reset - require copying FW file on USB stick.

Jerzeek, 2023-10-23:
Same problem here.

Pitel, 2023-10-24:
I've now successfully printed 2 small models without any crash! The key was to **disable binary gcode** in slicer.

samdurnxn, 2023-10-25:
No problem at all for me with binary gcode files but when I upload a file via prusaConnect wich have the same name as an existing file. The file is correctly overwrite on USB drive but when the printer want to start printing same BSOD.

I restart the printer, only change filename and everything is ok

chrishilbert, 2023-11-09:
Happens to me as well.  All my Minis will have this error at some point and require a restart.  It can happen from a wifi send OR from a selection on the screen.

I am trying to determine the cause.  It usually happens when I do back-to-back prints - the machine is still warm and cooling down.  I will do more tests to confirm.

matejus, 2023-11-12:
Identical exception ... after i loaded the file over PrusaConnect. It was the very first print over connect. It succeeded when I start the print of the same file again over connect. the same procedure with different file also finished without BSOD.

danopernis, 2024-06-03:
This is caused by dynamic memory allocation in heatshrink decompression library used in binary gcode. It is triggered anytime when the memory is fragmented and printer needs to open binary gcode. It is most prominent on MINI due to its smaller RAM, but may theoretically happen on any other printer. We are aware of the issue and tracking it internally as BFW-5023

The workaround is to disable binary gcode as suggested by @Pitel

Pitel, 2024-06-03:
I've kept using binary gcode, and actually hasn't seen any crash for a long time.

closed 2024-08-12 by github-actions[bot]