prusa3d/Prusa-Firmware-Buddy #3378: [BUG] Thumb drive stops reading mid-print

NerdOfEpic, 2023-10-22:
**Printer type** - XL 5 tool
**Printer firmware version** - 4.7.2
**Original or Custom firmware** - Original
**Optional upgrades** - None
**Printing from USB drive plugged directly into the printer**

**Describe the bug**
Mid-print, the printer appears to lose the ability to read from the USB thumb drive with the gcode on it.  The LCD interface has no indication that this has happened.  The print pauses, but USB icon is still present, the screen allows print resume.  Resuming tries to resume, heads for the print object, and immediately goes back to paused.  An attempt to capture a crash log when in this state results in a failure to save the crash dump.  Unplugging the thumb drive, plugging it back in, and clicking resume works properly for the print, and also fixes it for the crash dump scenario. 

**How to reproduce**
Print anything that takes a while from a thumb drive plugged directly into the printer.  I've had this issue pop up anywhere from 4 hours into a print to 9 hours into a print.  Once I discovered what was going on, I tried several thumb drives (different types, different capacities, different speeds) with the same result.  There doesn't seem to be any consistent amount of time before it happens, but once it does the user is forced to guess what is going on due to the lack of feedback.

**Expected behavior**
The firmware should detect that this issue has occurred and attempt to reset the USB system to continue reading the thumb drive on its own.  If for some reason this isn't possible, it should at least remove the USB icon in the upper right of the LCD and put up a message directing the user to unplug and replug the thumb drive before resuming.

**G-code**
Any gcode with a long print time will likely show this issue.  I've used the same gcode more than once and had it show this problem at vastly different times in the print.  This doesn't appear to be a gcode related issue.

MichaelKetchel, 2023-10-23:
Confirming, just experienced the exact same issue on a prusa XL with five tool heads using the Prusa provided USB. Print was uploaded via PrusaLink

Lagersuufer, 2023-10-29:
I've had the exact same problem using the 5.1.0-alpha2 code, both with the Prusa supplied USB and a brand new one. The failure appears to be random. I haven't had the problem on any single head prints, but my first 5 color print since upgrading has not been able to complete due to multiple failures within the same print.

Lagersuufer, 2023-11-08:
I've done at least six 9 hr prints using a single tool head with no problem whatsoever. Just started a 5 tool print and had the problem again while laying the first layer of the purge tower. Reset the USB drive, hit resume, and it started printing again. Seems to me like it may have something to do with the coding for 5 tool heads. My next print I will try it with just 4 active tools.

schuetzi99, 2023-11-10:
The same for me with 5.1.0-alpha2. So far it has declared two prints as garbage because a resume was not possible. And it happens with one tool head. This is really very annoying. VERY annoying!

MicroMan76, 2023-11-14:
I've had the same thing. XL 5 Head. I changed the USB drive from the Prusa to a Sandisk when it was happening frequently. It's been fine for a week or so and just happened again.
If I remove the drive and put it back it, it starts the print again directly. That seems to be the only way I noticed to un-freeze it. It just sits there with filament slowly oozing out of the head. 

I can see no way to reproduce it, the exact same code on the same USB drive will then work again.

I don't know if there is any connection, but the USB is very difficult to insert on my printer, well above what I would expect from a USB A socket. I wonder if there is some force on the plug which is creating an unstable connection? Does everyone else have this same high level of force required to insert the USB drive?

mikemccand, 2023-11-15:
> I don't know if there is any connection, but the USB is very difficult to insert on my printer, well above what I would expect from a USB

Same here, way too much friction.  Makes me nervous / causes me to flip the USB drive up and down like 7 times before I finally figure out which way!

I've had my XL 5T for a week now and have hit this same issue many times, both on 4.7.2 and 5.1.0-alpha2.  A long (8-14 hours) print will hit this a dozen times or more.  The printer just stops, keeping all heaters on.  When I re-seat the USB stick and click continue + resume, it resumes fine (though I have to remove big boogers if it's been too long), but one time, with 5.1.0-alph2, I got a bsod, and on resetting the printer, it failed to realize it was mid-print still, and I lost the print.  It was looking gorgeous too, sigh.

I've tried three different USB sticks, including the original Prusa USB stick, same issue.

Here's the one BSOD I hit:

[image]

Hmm, looking at the BSOD, maybe I was somehow running `5.1.0-alpha1`?  Why doesn't it say `-alpha2`?  Or maybe BSOD just doesn't differentiate those two versions.

MicroMan76, 2023-11-15:
I have another issue which might be related. When I power up the printer, the Ethernet light flashes on for a say 3 seconds and is off for 4-5 seconds. It just sits there doing this without the printer being accessible remotely. I have to remove the ethernet cable and re-insert it to stop the intermittent flashing and get comms. Is everyone who is having this issue also using pruslink/connect I wonder?

mikemccand, 2023-11-15:
It occurred to me that this may be heat related?  So I faced a small fan right at the back of the LCD controller/circuit board on the USB stick side during my 14 hour print that has already stopped at least a dozen times ... I'll report back.

mikemccand, 2023-11-15:
> Looking at that error... I have another issue which might be related. When I power up the printer, the Ethernet light flashes on for a say 3 seconds and is off for 4-5 seconds. It just sits there doing this without the printer being accessible remotely. I have to remove the ethernet cable and re-insert it to stop the intermittent flashing and get comms. Is everyone who is having this issue also using pruslink/connect with the ethernet cable I wonder?

Hmm I have not yet even tried ethernet cable -- I've tested PrusaLink, but over WiFi, and since I've been having all these USB hanging problems, I've stopped using PrusaLink and am just testing straight file copy on the USB stick, with straight gcode (non-binary).

mikemccand, 2023-11-15:
> It occurred to me that this may be heat related?

I now strongly suspect this is a heat problem.  My print, which stopped at least a dozen times in the first hour or two, has been running non-stop for ~2.5 hours now with no issue.  Now we maybe have a workaround!  Can anyone else confirm this sidesteps the issue for their multi-tool XL as well?

CogstateDStephens, 2023-11-15:
I don't have any extra fans blowing on anything, but it's presently about 15C/59F where I print.  If it were heat related, I feel like the ambient temp would have a pretty good shot at making it not happen for me.

MicroMan76, 2023-11-15:
A perfect 20.5C in the office all the time for mine.

slam878, 2023-11-17:
I have the same issue.  I've tried printing a file directly from the USB drive, or using Prusalink or Prusaconnect, but they all store the gcode on the USB drive.  I think this bug is made worse that the extruder will stop in place when the USB drive disconnected, and starts melting a hole into the print.

CZDanol, 2024-06-01:
Seems like the same issue as #3983 . Is this still happening on 6.0.0?

Majkl8383, 2024-06-15:
Hello guys,

finally it happend to me also. Yesterday I started print with TPU 98A using PrusaLink. 20h print started well and after about 6h thie error came up. There was no way how to restart print. So I went to my printer, sliced my file again this time I copied this file to absolutly clean Flash drive - different than before and started print again. Now after 14h of print again this error. This time using Prusa Connect I was able to restart it. But this bug is a serious problem, because of a lot of waste that could be created.

First crush yesterday was on FW 6.0.1, before starting new print i flashed brand ne FW 6.0.2 and then it happend again as describing below.

Using 5H Prusa XL.

Thanks for helping and solving this problem guys.

Majkl8383, 2024-06-15:
And now again that print I was able to restart stopped with same error after another 4 hours. Now without any chance to restart.

CZDanol, 2024-06-16:
@Majkl8383 you're talking about a different error I assume, not the tasks.c configAssert

Prusa-Support, 2024-11-30:
The firmware release [6.2.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1) includes overhauled USB media prefetch feature, which should drastically reduce the chance of "clogged buffer" hence "skipped lines/layers" or "frozen print".
A later firmware release [6.2.0-alpha2](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha2) proved to be a pretty stable firmware over a significant amount of testing hours, providing an effective correction of this erratic bug.
You can rely on the latest alpha release while the final firmware 6.2.0 with a few additional adjustments should follow in short.
 
Michele Moramarco
Prusa Research

closed 2024-11-30 by Prusa-Support