prusa3d/Prusa-Firmware-Buddy #3381: [BUG] XL dual-tool | 5.1.0a2 | false positive crash detection

Voxel3DPrinting, 2023-10-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [XL, dual tool, factory assembled]

**Printer firmware version** - [5.1.0a2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [none]

**USB drive or USB/Octoprint** - [Prusa connet]

**Describe the bug**
- A lot of false positive crash detections while printing with new IS firmware, sliced in PS2.6.1, 0,25 QUALITY profile. 11 false positive crash detections during 2hr print on medium sensitivity, High it will get higher, low it will get less but still present

**How to reproduce**
- Slice and print model with the 0,25 QUALITY profile from PS 2.6.1  with the 5.1.0a2 firmware on dual tool XL

**Expected behavior**
-  Prints fine without issues

**G-code**
 - Prefer not to share since it is a prototype of new product, but I replicated it with just a block of same size (60x110x150mm)

JeremiahGillis, 2023-11-17:
Same problem with my five head on the same firmware set to medium sensitivity. I see them when the nozzle probes the bed and also when the XY gantry moves.

JeremiahGillis, 2023-11-24:
I see the new released 5.1.0 firmware about crash detection and overly tight belts. I just want to comment that my printer is a five head fully assembled machine. The belts have been tightened by Prusa.

Rob65, 2024-03-02:
Still an issue on a newly delivered printer (dated 22.02.2024) with the latest (5.1.2) firmware

Felteshobbies, 2024-03-23:
Same here with Input Shaper Default Profile on FW 5.1.2+13478 and 0.15 Speed.
Seems, it happened on the fast non printing xy moves and i try to reduce the max value 7000 to 5000.

closed 2024-06-26 by github-actions[bot]