prusa3d/Prusa-Firmware-Buddy #3384: [Bug] Prusa Mini Fails to register to prusa connect

Reesesben, 2023-10-22:
I did a quick search to see if this had been added before but I don't see it sorry if I'm wrong

**Printer type** - [MINI]

**Printer firmware version** - [5.1.0-alpha2+13126]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor, ESP-01s WiFi chip]

**USB drive or USB/Octoprint**
  USB/na

**Describe the bug**
  When trying to register with prusa connect on the prusa mini it fails every time. 
[image]

**How to reproduce**
  Using a ESP-01s WiFi chip flash the WiFi firmware and assign credentials, confirm WiFi is working by establishing a connection via the listed IP. Go to prusa connect settings and select add printer to connect.

**Expected behavior**
  Printer should display a registration code for me to give to the prusa connect website 

**G-code**
N/A

**Crash dump file**
No crash file given by printer sorry
**Video**
https://youtu.be/Ru6PVdLKYas

wilburb, 2023-10-24:
@Reesesben I contacted support about this, as both my XL and Mini were showing the same symptoms as you describe. Support informed me they had been having issues with Prusa Connect lately. They were unable to provide me with an estimate of when it might be fixed. They did suggest I try add it again later or wait a few days and try again. I happened to try it once a week or so later and both successfully added with no further changes on my end.

I assume you may be hitting this same issue, and waiting/trying a couple times a day may be your best bet. I wish they’d have a status page, as I understand it’s beta software, but it would still be nice to know if it’s on my end, or a known issue on their end when things fail to work as expected.

petous22, 2023-10-24:
I have the same problem with my MK4 after today's firmware update to version 5.1.0-alpha2.

Reesesben, 2023-10-28:
> @Reesesben I contacted support about this, as both my XL and Mini were showing the same symptoms as you describe. Support informed me they had been having issues with Prusa Connect lately. They were [...]

Sorry I should have added that i've been trying over the course of a month, (when they released the first mini + firmware update) I try once or twice a week and get the same result.

Reesesben, 2023-10-28:
Update:
I've found a temporary work around but this issue still needs addressed.

Step 1
in network settings load the .ini file to load credentials from the printer to the usb using the "Create Credential" Button under Settings/Network/WiFi

Step 2
Add your SSID and Password.

Step 3
Add these lines to the file

[service::connect]
hostname = buddy-a.connect.prusa3d.com
tls = True
port = 443

Step 4
Load the credentials to the printer. Mine was autodetecting them but if yours doesn't go to Settings/network/wifi and press "Load Credential"

Step 5
Wait for the WiFi symbol to re-appear after it does hit the reset button on the printer next to the knob.

Step 6
Go to Settings/Network/Prusa Connect and hit the button that says "Register printer to connect" It should work now if it doesn't give you a code try reflashing the ini file and rebooting the printer again. 

Step 7
Connect will load the printer but the prusa mini won't actually connect. Go to the prusa connect page and select the Settings tab, under the LAN settings portion select "Download settings" attach the file you receive to the USB drive

Step 8
Plug the drive back in the printer and go to Settings/Network/Prusa Connect and select "Load settings"

Step 9
Wait for the WiFi to reconnect and reboot the printer. 

Step 10
Refresh your browser you have connect.prusa3d.com open in and see if it worked. 

My connection is unreliable at best but the printer recognizes prusa connect as a thing now and the website acknowledges my printer so its a step in the right direction.

closed 2023-11-21 by Reesesben