prusa3d/Prusa-Firmware-Buddy #3385: [BUG] Prusa XL on 5.1.0-alpha 2 firmware lost usb drive midprint

mogorman, 2023-10-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Prusa XL 2 tool assembled 

**Printer firmware version** - 5.1.0-alpha2

**Original or Custom firmware** - original

**USB drive or USB/Octoprint**
Printed from prusaslicer to prusa xl directly to usb drive

**Describe the bug**
I was printing a larger print, 100 mb and at 4% competition it lost the usb drive. reinserting it and telling it to try again a few times caused it to move ahead 

**Expected behavior**
Print without issue.

**G-code**
would not let me attach the gcode to the github issue.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

mogorman, 2023-10-22:
happened again at 37% on same print

CommanderRedYT, 2023-10-23:
Try wiggling the USB drive the port while printing. Maybe it is making faulty connections

mogorman, 2023-10-23:
i dont think it was that as hasnt been a problem before upgrading and i never touch it, i do all my prints via the network. happened a third time at 55% on that print.  seems unlikely that its 3 times in one print on alpha firmware, and never been a problem before that.

jacobfrasca, 2023-10-28:
Had same issue but at the end of my print. Was using Prusa Connect. on 5.1.0- alpha 2 sliced in PS 2.7.0 alpha 1
[dump_buddy.bin.dmp](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13194099/dump_buddy.bin.dmp)

CommanderRedYT, 2023-10-28:
Oh wow, I don't have that much experience with the prusa firmware in particular but still from experience, watchdog trigger sounds like it crashed because some function / task took way too long. Are you also connected to wifi / Ethernet?

jacobfrasca, 2023-10-28:
I’m not sure, but the software indicated before this that my usb drive was removed and that I need to reinsert it. That’s why I uploaded the dump cause honestly it’s way out of my expertise.

closed 2024-06-26 by github-actions[bot]