prusa3d/Prusa-Firmware-Buddy #3386: [BUG] BSOD when printing objects serially (Print Individual Objects).

ex-nerd, 2023-10-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

Would be nice if it had somewhere to do a purge/wipe

**Printer type** - XL 2-tool

**Printer firmware version** - 5.0.1-alpha

**Original or Custom firmware** - Original (alpha)

**Optional upgrades** - 2-tool

**USB drive or USB/Octoprint**

USB/wifi

**Describe the bug**

Printer restarts and then BSODs while trying to printing multiple parts in serial.

**How to reproduce**

- Send print job to printer
  - I've experienced this with a couple different print jobs, and "serial printing" is the single commonality.
- Printer finished first part and as it was moving toward where it would start the next, it restarts. After the boot finishes, it shows the BSOD screen.
  - Note that with this project "one part" in the serial print is actually 2 physical objects. Other projects have had similar failures with single-object "parts."
  - I've had other instances where it finishes multiple parts before exhibiting the same crash, but the file attached to this ticket so far always crashed after the first part, after moving away from that part (X/Y move successful) and right before what I assume is the Z-down move to start printing the second set.

[image]

```
precise_stepping. cpp: 447
Negative_step time: -938, flags: 3896 task: IDLE
SP: 2002fd60
LR: 0803bf1b
PC: 0803bf1a
CPUID: 10004a48
ICSR_:0004019b
VTOR_: 00000001
AIRCR: 10004b50
SCR_:10004a44
SHCSR: 2002fda4
HFSR_: 081a162d
DFSR_: 2002ff04
AFSR_: 20019748
DFR_:2002Fd60
ADR_:0803bf1b
CPACR: 00010000
bot:2000df74
top: 2000ddc8
stack: 2001d28c a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5
...
```

**Expected behavior**

I expect the print to succeed

**G-code** + **Crash dump file** + **3mf project**

Attached .zip contains bgcode, crash dump, and 3mf project:
[prusa-xl-crash.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13064552/prusa-xl-crash.zip)

bstansell, 2023-10-25:
Hit the same issue on my MINI+.  Here's a picture and the `dump_buddy.bin` file in the zip.

[image]

[crash.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13169395/crash.zip)

danopernis, 2024-06-03:
This is fixed since v5.2.0 via https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/a1248c14b6f2394916e76562b17ae75d4f18fdc4

closed 2024-06-03 by danopernis