prusa3d/Prusa-Firmware-Buddy #3387: [BUG] Z Stepper Motors won't shut off - remain at 40°C

IBNobody, 2023-10-22:
**Printer type** - MK4

**Printer firmware version** - 5.0.0

**Original or Custom firmware** - Original

**Optional upgrades** - None

**USB drive or USB/Octoprint** - USB

**Describe the bug**
  Both Z Stepper Motors are not shutting off when idle. They are constantly energized and are warm to the touch (40°C), even hours after a print. The X, Y, and extruder stepper motors are all room temperature.

**How to reproduce**
  Print a model, wait an hour.

**Expected behavior**
  The Z stepper motors should be powered off and sit at room temperature.

**G-code** - N/A

**Crash dump file** - N/A

**Video** - I will post images taken with my thermal camera later.

IBNobody, 2023-10-22:
I will be monitoring this to try and dial in the exact moment or mechanism that causes the steppers to remain energized. My next test will be a Z axis movement.

faunris, 2024-02-12:
The same problem. Just power off and on the printer. 
MK4 firmware 5.1.3

bkerler, 2024-02-16:
As stated by the Prusa devs, this is intentional and needed for the MK3.5 and MK4, so it is no bug. Please close this issue.

Gl-Mk4, 2024-06-11:
After switching off the MK4 for two weeks, the Z axis still is not down at the bed. So when a print is **not paused** but completed, I see no reason for burning 11 Watts for hours! 
Furthermore, at least the existing menu option "Disable motors" should do so.

danopernis, 2024-06-11:
This duplicates https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3339 and will be closed.

closed 2024-06-11 by danopernis