prusa3d/Prusa-Firmware-Buddy #3393: [BUG] XL Multi Tool - No nozzle temperature after power loss recovery

Dorito203, 2023-10-23:
**Printer type**
XL Multi Tool

**Printer firmware version**
Firmware 4.7.2
Bootloader 2.0.8

**Original or Custom firmware**
Original

USB drive

**Describe the bug**
  I tested the power loss recovery feature by turning off the power during a print (single color print). The power was off for about 10 seconds, then I turned it back on. The printer continued printing but did not set a nozzle temperature.

**How to reproduce**
  I have not yet tried to reproduce this error.

**Expected behavior**
  The printer should return to the designated temperature.

**G-code**
 The gcode is unfortunately not existing anymore.

**Crash dump file**
 No crash dump file

cechurat, 2024-03-18:
**Printer type**
XL Multi Tool

**Printer firmware version**
Firmware 5.1.2+13478
Bootloader 2.3.4

**Original or Custom firmware**
Original

I can confirm the above-mentioned issue. After powerless recovery, there was no nozzle temperature so the print was destroyed.

Thank you for looking into it.

danopernis, 2024-06-03:
Created internal ticket BFW-5621

CZDanol, 2024-11-27:
Hey, this should get fixed in the next 6.2 release candidate.

CZDanol, 2025-01-09:
Hello, we've just released 6.2.0-RC1 where the issue should be fixed: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-RC1

I'm thus closing the issue. If the problem re-appears, please notify us :)

closed 2025-01-09 by CZDanol