prusa3d/Prusa-Firmware-Buddy #3394: [BUG] <Enter comprehensive title>

Dorito203, 2023-10-23:
**Printer type**
XL Multi Tool

**Printer firmware version**
Firmware 4.7.2
Bootloader 2.0.8

**Original or Custom firmware**
Original

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
The printer simply stopped at 96% and didn't move any more. The temperatures were still set. The remaining print time read N/A.
I pressed pause print, it did, then I resumed printing, it moved back to the model and immediately paused printing again. I tried this two or three times and then stopped the print since the model was almost done anyway.
The error has not reappeared since then.

**How to reproduce**
 I don't know if the printer would repeat this error with the gcode.

**G-code**
[fehlerhaftTeller Tief_0.6n_0.35mm_PLA_XL_7h22m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13072216/fehlerhaftTeller.Tief_0.6n_0.35mm_PLA_XL_7h22m.gcode.zip)

**Crash dump file**
  No crash dump file

**Video**
[20231018_234359.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13072228/20231018_234359.zip)

rcmaniac25, 2023-10-26:
Myself and some others have had this issue. I hit it on my XL single tool. Luckily, it happened like 4h into a 49h print... but still was not happy. Support asked if it happened again once reprinted... has not happened since that one time.

jeffyl, 2023-11-03:
I just got my 5-toolhead XL and have been somewhat consistently having this problem with the same unable to resume printing. In my latest attempt I was able to resume the print by removing the USB drive and plugging it back in before attempting to resume.

JeremiahGillis, 2023-11-17:
I've had this same issue with `5.1.0-alpha2`. I've had my stop right when the nozzle is touching the bed, and also when the nozzle was over the model touching it. I had to cut away the oozing before continuing the print.

[image]

I paused the print, then restarted, and got `USB drive or file error, ...`, so I pulled the flash drive out, inserted the flash drive, then was able to resume the print. I was used the included Prusa branded flash drive.

tg73, 2024-02-18:
I'm seeing a single hang of 20-30 seconds a few seconds into the first layer. I've seen it twice now. It gets going again of its own accord. No unusual messages on the display.

tg73, 2024-06-06:
Waiting for Prusa to respond.

Prusa-Support, 2024-09-07:
Thanks for your feedback.

This issue may have multiple causes so it is probably related to the existing issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3378, duplicated by https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3664, and again related to later issues https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3682 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3983 which dragged more attention from the community.

In other words, this issue may be observed from several perspectives but may potentially have something to do with data flow & clog to various extents, and I recommend giving the order issues a read.
A common element in all of these cases is for sure their erratic nature, which makes them hardly reproducible (if reproducible at all) and difficult to work with.
However, data flow optimization takes place every few firmware releases following a natural course of firmware optimization, carried side by side with several other firmware improvements and new feature implementations.
In other words, a lot of progress has been made but this is a perpetual work of optimization.

We got several reports from different perspectives but generally, these remain sporadical episodes affecting a relatively small percentage of printers (and prints).

Would these episodes occur so frequently as to have a significant impact on your regular printing workflow, please...

- Consider reducing background processes.
  - For example, disabling the print preview, or avoiding printing with Gcode streaming...
  This would be a temporary experimental condition for debugging reasons.
- Consider comparing the results with other USB drives.
  - Max 32GB, with filesystem FAT32, ideally type USB 1.0.
- Reach out to our Customer Support to discuss your individual case and discuss tailor-made solutions.
  - https://help.prusa3d.com/article/customer-support_2287
  
Michele Moramarco
Prusa Research

CZDanol, 2024-09-13:
We've just released a 6.2.0-alpha1 FW which hopefully fixes the USB errors. If there are new problems, please create a new issue.

Closing this one, because it contains reports from quite old firmwares.

closed 2024-09-13 by CZDanol