prusa3d/Prusa-Firmware-Buddy #3396: [BUG] No progress screen - MK4 5.1.0-alpha2

xcybplx, 2023-10-23:
**Printer type** - [MK4]

**Printer firmware version** - [5.1.0-alpha2.]

**Original or Custom firmware** - [Original/Custom]

**USB drive or USB/Octoprint**
  USB + PrusaConnect + PrusaSlicer 2.7.1 Alpha

**Describe the bug**
After starting the print, the print progress screen is not showing. All settings are correct.
[image]
[image]

**Expected behavior**
Please fix the issue

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
[Wypelnienia.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13073185/Wypelnienia.zip)

xcybplx, 2023-10-23:
For some strange reason, it started to work at 43% and now even if I will go to the settings, it works as expected.
[image]

wgentine, 2023-10-24:
Showed up later here too...

closed 2024-06-26 by github-actions[bot]