prusa3d/Prusa-Firmware-Buddy #3399: [BUG] BSOD with BGCODE from Prusa Connect

chrishilbert, 2023-10-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** -5.1.0-alpha2+13126

**Original or Custom firmware** - Original

**Optional upgrades** - N/A

**USB drive or USB/Octoprint**
  USB/PrusaConnect

**Describe the bug**
I got BSODs on multiple printers sending BGCODE to printers from PrusaConnect

**How to reproduce**

Uploaded BGCODE to PrusaConnect Files,
Clicked Start Print
Select Printer
Printer will download BGCODE then immediately BSOD.

**Expected behavior**
Print should start and no BSOD.

NightlyH, 2023-10-29:
I too have this issue

Freycraft, 2023-10-30:
I have this bug too

MihRei, 2023-10-30:
I have this too. I tried just to upload and print option, bg code, through Prusa Slicer to Prusa Connect and then start printing. when the printing starts, bsod. Then I tried through Prusa Slicer just to upload to Prusa Connect and after that to start the printing in Prusa Connect web interface, same problem after the printing process tries to start. 

I will let here the file that the printer have 

[dump_buddy.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13206437/dump_buddy.zip)
generated.

danopernis, 2024-06-03:
This is a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3375

closed 2024-06-03 by danopernis