prusa3d/Prusa-Firmware-Buddy #3400: [BUG] USB drive or file error

wgentine, 2023-10-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK4]

**Printer firmware version** - [5.1.0-alpha2]

**Original or Custom firmware** - [Original]

**USB drive or USB/Octoprint**
  USB Drive+PrusaLink

**Describe the bug**
  During the print with prusaslicer 2.7.1 alpha, printer stopped mid print saying: USB drive or file error, the print is now paused. Reconnect the drive. However the USB drive is there and nothing has changed. May be something related to the new binary g-code/streaming.

codingcatgirl, 2023-10-24:
Check out the "known issues" section of the firmware, this issue is listed right there.

wgentine, 2023-10-24:
> Check out the "known issues" section of the firmware, this issue is listed right there.

absolutely missed that!

rwhitenack, 2024-03-13:
Can someone help me with the location of the above issue. Is I've looked and looked and can't find "Known issues" in the firmware section. Is the "firmware" section from the prusa3d website? On the github forum? In the downloaded file of the firmware? I've looked in all three and not found anything on "known issues". I have looked in the "[Issues](https://github.com/prusa3d/Prusa-Firmware/issues)" section of Prusa-Firmware on Github but couldn't find anything related to the issue in the subject of this post.

wldthng842, 2024-03-20:
Running into the same issue here mid print. USB is just fine, PC recognizes it and everything.  Were they any solutions besides formatting and starting over?

codingcatgirl, 2024-03-20:
@rwhitenack i'm referring to the known issue section of the exact firmware version mentioned in this issue's title: 5.1.0-alpha2. It's right here: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha2

@wldthng842 If you are indeed running into this exact same issue, which refers to the 5.1.0-alpha2 firmware, you would probably want to update your firmware to the current version.

codingcatgirl, 2024-06-26:
The fact that according to the latest firmware release **yesterday**, Prusa is **confirmed to be working on this issue right now** demonstrates **perfectly** why stalebot is so, pardon my words, shit.

AndorraM, 2024-06-28:
I got the USB error twice today, even after updating to 6.0.3 on my 5TH XL. Fixed it by replacing the file with a freshly sliced file from my computer. But it's rather frustrating that the firmware upgrade did not fix or change anything from what I can tell.

m10d, 2024-06-30:
@danopernis @josefprusa @Prusa-Support would you please update your use of stalebot: i do not believe you're using it correctly. 

- Please add the configuratiom (below) for excemptions if its not already in place.
    - directly supported by yhe stale action:
https://github.com/actions/stale?tab=readme-ov-file#exempt-issue-labels
- Please make sure your admins have a process in place to actually mark real, accepted issues -at least those that are expected to get worked!- as not to be stale-closed
    - one suggestion: write an `action` of your own that applies one of these exempt labels to any issue with over X distinct commenters or Y likes (by definition distinct users) - of course these could still become stale but such an issue with lots of impact presumably already has one or more authors engaged, so this shouldn't be mich of a burden?

I offer this issue as an example; i am sure you can understand the frustration of users when a *real* bug is closed, just because they are patiently waiting for a fix.  
    - (If its not consensus that this issue should have been excempted PLEASE follow up here - i would really like to understand that rational)

Further, as a software engineer i fully understand & support the necessity of OSS projects using the stale-bot; with the amount of drive-by or not-reproducible issues, it is just the only way (a submitter has the obligation to provide reasonable follow up - but many just don't :( HOWEVER obviously many users don't share this background and Prusa's image would benefit by hyperlinking to a thoughtful explaination/justification directly from the stale-action close message.

m10d, 2024-07-12:
@danopernis - looks like you disagree with the points i'm making / actions i am requesting; (assuming you are indeed associated with prusa) please elaborate.

danopernis, 2024-07-12:
Sure @m10d let me react. First of all, thank you for recognizing the importance of having the stalebot in the first place. That is something that we could both agree to.

The disagreement stems from this: I think the stalebot as it is configured right now is sufficient to deal with most of the situations. Sure, it may mark some issues to be closed when they indeed shouldn't be. That's why there is a time period for folks to react to the stalebot. I am actively looking at those issues and if the issue is still relevant, I at least assign it to somebody to prevent the stalebot from closing it. Some of the them might still slip, either because of error on my side or because folks don't react in time. We can still reopen such issue if it is popular enough. In words of Agile Manifesto,

> Individuals and interactions over processes and tools

Feel free to suggest such issues. Unlike stalebot, we are people not machines and can always correct any mistakes, it just requires some communication.

As for the user frustration, I get that, but in the end, GitHub is intended mostly for people with some technological background. Most of such users understand the importance of stalebot and react in professional way, usually just by commenting on the issue to prevent closing. It boils down to not taking it personally, that's all.

All in all, I think the process is pretty good as it is and don't see any need to change it. FWIW I considered your suggestion to better explain our position in the stalebot automatic response and discussed it with colleagues. But in the end, no matter what you do, haters are still gonna hate it, so what is the point of changing the automatic response?

m10d, 2024-07-31:
It sure would be nice if prusa would add more detail about what root-cause analysis suggests is at fault here. From all the various forums (github, prusa forum, reddit) it sure seems this is EMI related due to the long unshielded ribbon cable housing usb D+/D-, SPI, and i2c. I don't have any experience with USB, but spi and i2c are (much) more commonly employed on one board or board-to-board with very short cables only. Let alone running adjacent to motors.

Prusa, would you please consider:

- weighing in on what issue should track this, so we're not looking at mulitple, in multiple repos
- provide a summary of your testing, in case some very easy workarounds might bear some fruit:
    - separating cable traces, wrapping/shielding w/ ground on single end
    - ferrite / choke (what spec, containing which lines of the ribbon)
    - outright replacing the ribbon with something else. Note: [Pinnouts incl' xBuddy and LCD boards](https://www.prusa3d.com/downloads/Electronics_drawings/MK4_electronics_schematics.zip), termination of the lcd cable with [TE Micro-MaTch](https://www.digikey.com/en/products/detail/te-connectivity-amp-connectors/9-215083-4/4486392) (via [forum](https://forum.prusa3d.com/forum/user-mods-enclosures-nozzles/mk4-display-cable-to-short-for-the-prusa-original-enclosure/#post-657125))

Prusa-Support, 2024-09-25:
Not sure if this issue is easily reproducible on our side but it may be overlapping or duplicating other similar issues.
I wonder if this issue has actually to do with the known USB disconnect problem mentioned in recent release notes and potentially fixed with the overhauled USB media prefetch feature of [FW 6.2.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.2.0-alpha1).
Please give it a try to help with testing if you can, and the stable release should follow sometime soon.

Michele Moramarco
Prusa Research

closed 2024-12-10 by github-actions[bot]