prusa3d/Prusa-Firmware-Buddy #3401: [BUG] Don't heat up unused XL extruders to 170C

mdjurfeldt, 2023-10-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

5-tool XL

**Printer firmware version** - [e.g. 4.0.5, ...]

5.1.0-alpha2

**Original or Custom firmware** - [Original/Custom]

Original

**Describe the bug**

When I do a multimaterial print involving less than five extruders, the unused extruders are heated up to 170C. This causes unnecessary energy consumption.

**How to reproduce**
  Please describe steps to reproduce the behavior.

Perform a multimaterial print with less than five extruders. Look at the temperature of the unused extruders.

**Expected behavior**
  A clear and concise description of what you expected to happen.

I expect the unused extruders to have room temperature.

thijstriemstra, 2023-11-13:
> When I do a multimaterial print involving less than five extruders, the unused extruders are heated up to 170C.

😱 can't believe this wasn't tested by prusa. What about single material prints, I hope the remaining 4 aren't heated up?

mdjurfeldt, 2023-11-14:
An additional piece of information: In several prints, extruders not participating in a print have, correctly, not been heated up. It might be that they are wrongly heated up only if they have filament loaded.

jhnphm, 2024-06-20:
I've observed this as well when using extruders 2-4, with 5 joined to 4 and extruder 1 unused. Extruder 1 remains at 170. Extruder 5 is fully heated as described in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3874

Razor412, 2024-12-12:
This is still happening with 6.2.0-alpha2+8702. 

> I've observed this as well when using extruders 2-4, with 5 joined to 4 and extruder 1 unused. Extruder 1 remains at 170. Extruder 5 is fully heated as described in #3874

Also, #3874 is still a thing.

CZDanol, 2024-12-12:
Hello, if this is what I think it is, it should be under our internal ticket BFW-5996 and should be fixed in the next 6.2.0 candidate.

closed 2025-12-09 by Prusa-Support