prusa3d/Prusa-Firmware-Buddy #3402: [BUG] Stringing XL 5T

the0bone, 2023-10-24:
**Printer type** - XL 5T

**Printer firmware version** - 5.1.0-alpha2 as well 4.7.2

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
  Massive Stringing is still present at the 5T. Files sliced with PS 2.6.1 or 2.7.0 a1. No difference.
  All Filament is dry. Prusament, Filament PM and Extrudr.

**Pictures**
FW 4.7.2 with PS 2.6.1
 [image] 
FW 5.1.0-a2 with PS 2.6.1
[image]
FW 5.1.0-a2 with PS 2.7.0-a1
[image]

the0bone, 2023-10-25:
[Testprint 3mf.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13165757/Testprint.3mf.zip)
[Testprint bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13165758/Testprint.bgcode.zip)

Noeliu1997, 2023-10-26:
Please try this file with 4 PLA on tools 2-5. I didn't use n1 for convenience since I had PETG loaded and was sending this through connect. 
Its really slow, since I forgot to turn off the cooldown on the tools, but 27minutes for the test is worth it

This way we can really start to see if its something on the slicing or the machine itself.
Please repost back with results so we compare.

I used Prusament. Please make sure you use dried and quality filament
[image]1st test Zusammengefuhrt_0.6n_0.32mm_PLA_XL_27m.bgcode
[1st test Zusammengefuhrt_0.6n_0.32mm_PLA_XL_27m.bgcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13173130/1st.test.Zusammengefuhrt_0.6n_0.32mm_PLA_XL_27m.bgcode.zip)

the0bone, 2023-10-27:
Tuned the Settings a lot and found a almost perfect way for a lot of PLA and PLA+ Filaments.
Even increased the speed.
[Generic_PLA @XL 0.6.ini.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/13192993/Generic_PLA.%40XL.0.6.ini.txt)

garethky, 2023-10-28:
Can you do a diff of the profiles so we can understand what parameters you changed and how?

the0bone, 2023-10-28:
Decreased temp 215 to 210.
215 worked with most filaments I tested with below settings but Filament PM white was not good. 210 worked better.

Overwriting printer settings via Filament:
Deretraction Speed from 20 to 30.
Retract amount before wipe from 80% to 0%
Retraction Length from 0.7 to 0.8
Lift Z from 0.2 to 0.4

and just because the printer is able:
Max vol speed from 15.5 to 21.

the0bone, 2023-11-11:
Thanks Prusa for fixing it.

PrusaSlicer 2.7.0 beta is tackling that issue. By several steps they got it fixed. The major one that you can see during printing, is that the nozzle purge a bit into the tower before doing the Toolchange. That is similar to the G-code change, that was published on YouTube before.

[image]

closed 2023-11-11 by the0bone